PX4/PX4-Autopilot #3840: Optical Flow with PX4

Linja0314, 2016-02-24:
Hi guys,

It's my first time use "px4flow + pixhawk + PX4 Flight Stack",
and I have experience for "px4flow + pixhawk + APM Flight Stack".
But now it's totally different.
Does someone have successful experience can share with me? Thanks a lot.

LorenzMeier, 2016-02-24:
Did you follow the instructions here?
http://dev.px4.io/flow_lidar_inav_setup.html

LorenzMeier, 2016-02-24:
Also make sure to flash the flow module with QGroundControl and the default binary and nothing else.

Linja0314, 2016-02-25:
Many thx for your reply.
Now I follow three web page's setup
https://pixhawk.org/modules/px4flow#orientation
http://dev.px4.io/flow_lidar_inav_setup.html
https://pixhawk.org/firmware/parameters

Q1 Are the parameters from "https://pixhawk.org/modules/px4flow#orientation" can be utilized tp control both PX4Flow and LIDAR-Lite modules ?

I find those parameters relative to Optical flow :
LPE_FLW_XY
INAV_W_XY_FLOW
INAV_W_GPS_FLOW
INAV_FLOW_K
INAV_FLOW_Q_MIN
I think we can use the default value of above parameters, first. AM I right?

Q2 This problem happened on APM Flight Stack.
My sonar is MB1240 XL-MaxSonar®-EZ4™, refer to this page; http://www.maxbotix.com/Ultrasonic_Sensors/MB1240.htm
I think the pin definition of MB1240 XL-MaxSonar®-EZ4™'s is the same with HRLV-EZ4.(except Pin 1 ,but still pull High)

Q3 We only make px4flow funtion work with APM flight stack(FW version is APM 3.3.2 rc2), do you ever succesfully utilize px4flow with PX4 flight stack and which FW version is it ?

Q4 When we utilized px4flow with APM 3.3.2 rc2 firmware, we seperate px4flow and sonar sensor; px4flow communicates via I2C and sonar sensor communicates via analog port. 
Do you ever let sonar sensor attached on px4flow, and update optical flow info and height info to Pixhawk via I2C ? (We failed to do this.)

Linja0314, 2016-03-11:
Hi guys,it's me again.

I can read some data from pixhawk if  i move the drone
http://imgur.com/9Mt785S

But I still don't know how to work with flow ,it's LOITER mode?
My pixhawk's Firmware is PX4 V1.2.0 and PX4flow's Firmware is use to flash the px4flow module with QGroundControl'ss stable version.

jgoppert, 2016-03-18:
@Linja0314 Try using the lpe esitmator, the position hold with flow is working well with it, I just flew it last night. You can build with:

```bash
make px4fmu-v2_lpe
```

You will need to set INAV_ENABLED = 0 after flashing and reboot. Then LPE should be running.

Linja0314, 2016-03-30:
@jgoppert  Sorry for late reply.
I already use  make px4fmu-v2_lpe  and i can switch POSCTL
When I set INAV_ENABLED = 0 ,I think it's not work with px4flow
And it's my video link
https://youtu.be/aaPsQVbEHSg
Could you please give me some advice?

jgoppert, 2016-03-30:
@Linja0314 we have been doing a lot of improvements here: https://github.com/PX4/Firmware/pull/4036
Pleased try that branch.

jgoppert, 2016-03-30:
Also, the flow only works well from about 30 cm to 3 m.

closed 2016-05-03 by tumbili