PX4/PX4-Autopilot #3841: PX4-ROS-Package - broken?

TobiMiller, 2016-02-24:
Hi,

i am trying to compile the newest firmware as ros package. it runs well, but ros doesnt find the package after install. the workspace is ready to use because the other packages build in this catkin_ws are working properly. bash.rc is working. 

i am now running an old version from "px4ros". this works.

is ros capabilty broken actually?

LorenzMeier, 2016-02-27:
@TobiMiller Could you describe what you are trying to achieve with it? We have largely evolved our ROS-facing infrastructure over the last couple of months.

TobiMiller, 2016-03-16:
its fixed. the newest px4ros repository is working. 

but the mavlink problem still exists. 
1.

> roslaunch px4 gazebo_ardrone_empty_world.launch

working without erros, gazebo is up and running

2. 
> roslaunch mavros px4.launch fcu_url:="udp://:14540@127.0.0.1:14557" 

=> connects, but it stops at

> [ INFO] [1458132618.198629927, 116.020000000]: MAVROS started. MY ID 1.240, TARGET ID 1.1

any suggestions? thats the reason why qgroundcontrol dont connect

closed 2017-04-02 by mhkabir