PX4/PX4-Autopilot #3843: VTOL: standard VTOL climbs a lot during forward transition

AndreasAntener, 2016-02-24:
Effect gets bigger in strong head wind.
Could potentially be improved by tighter MC altitude tuning. Might need an extra mode in the position or vtol controller to support the faster reaction to sudden wing lift.

AndreasAntener, 2016-03-05:
More logs from Sander showing the in #3926

AndreasAntener, 2016-03-05:
@sanderux @tumbili 
My guess is that the pusher together with the quad holding attitude create more lift than the mc pos controller compensates, or we don't control Z velocity anymore during the transition.
Later could have something todo with the following control flags set to false:
```
control_mode.flag_control_position_enabled
control_mode.flag_control_velocity_enabled
```

sanderux, 2016-03-05:
[image]

I think the pitch is being pushed up by the pusher and the quad compensation is too late

sanderux, 2016-03-05:
[image]

on the other hand you might be right, the quad doesnt seem to throttle down much when altitude increases

AndreasAntener, 2016-03-05:
yea I'm not yet sure. I got suspicious when I looked at the local position velocity setpoints during the transition. They seem to be reset. Didn't spot something obvious in the controller though.

AndreasAntener, 2016-03-14:
Fixed on master in a62046625b234dcc71279d036b6a70530c67c030

closed 2016-03-14 by AndreasAntener