PX4/PX4-Autopilot #3847: Extend FrSky Telemetry to match existing solutions

LorenzMeier, 2016-02-24:
@kd0aij @pkocmoud After collecting some user feedback they are looking for a match to this solution:
https://github.com/Clooney82/MavLink_FrSkySPort/tree/s-c-l-v-rc-opentx2.1

LorenzMeier, 2016-02-24:
After further investigation the main thing missing is GPS status.

careyer, 2016-03-10:
IMPORTANT: The above mentioned telemetry solution is widely spread among FrSky Taranis users and is currently the most comprehensive telemetry solution . It would be highly favourable if the Pixracers output on the FrSky-Port would match the output of the aforementioned Teensy solution (i.e.: outputing the same parameters and writing them to the same FrSky "sensors" - This is a mandatory requirement so that the LUA Scripts on Taranis stay compatible). -- THANKS in ADVANCE!

[image]

hsteinhaus, 2016-03-13:
I just would like to mention that APM's FrSky telemetry driver has mostly adopted to PX4's wire format. Changing the semantics of fields would break this compatibility and make it difficult to switch between both flight stacks. 

Are there any chances to convince the authors of the MavLink converter to switch to the PX4/Ardupilot field definitions instead?

I am using both FrSky telemetry interface solutions, the Teensy/Mavlink converter and a simple passive cable with just two 1k resistors as well as both flight stacks, so my only wish is compatibility. But just replacing one compatibility problem by another one does not really help :-)

kd0aij, 2016-04-23:
discussion at https://github.com/PX4/Firmware/pull/3762

closed 2016-04-23 by kd0aij