PX4/PX4-Autopilot #3858: Need to split the acro switch into MC acro and FW stab

AndreasAntener, 2016-02-25:
Currently the same switch is used for both modes. This is bad for VTOL if you fly in manual but switch fixed-wing stabilized on later. As soon as you transition back to multicopter you'll be in acro. Happened to me a few times already, it's challenging ;)

LorenzMeier, 2016-02-25:
:dash: Sounds fun! We need to have an informed discussion about flight modes anyway.

tubeme, 2016-02-27:
This was confusing for me too so that's why we use the FW perm stab in the VTOL. If it is with one switch at least it should work in the same direction in both modes...

LorenzMeier, 2016-07-29:
Release 1.5 will handle flight mode cleanup.

kd0aij, 2016-08-09:
bugfix here: https://github.com/PX4/Firmware/pull/5288
Looks to me like the transition logic was screwed up in commander.

AndreasAntener, 2017-02-18:
Fixed in #6567

closed 2017-02-18 by AndreasAntener

dagar, 2017-02-18:
@AndreasAntener could we now get rid of fw_permanent_stab or does it exist for a different reason?

AndreasAntener, 2017-02-19:
Only works if you assign the manual switch now in the old style, because that makes stabilized the default (btw, which is also the default for us in MC to be consistent).