PX4/PX4-Autopilot #3860: Gazebo standard vtol simulation broken

DroneDude123, 2016-02-25:
The gazebo simulation is broken for the standard vtol simulation. The aircraft will not transition properly.

Steps to reproduce:
1. make posix_sitl_default gazebo_standard_vtol
2. In qgroundcontrol, load the mission at the bottom of this message. This mission consists of the following wavpoints (Takeoff, Wavpoint, Vtol Transition (Plane), Wavpoint). 
3. Switch to mission mode in QGroundControl.
4. Arm aircraft.

The aircraft will take off just fine, but it will pitch up wildly during transition and go unstable.

`{
    "MAV_AUTOPILOT": 12,
    "groundStation": "QGroundControl",
    "items": [
        {
            "autoContinue": true,
            "command": 22,
            "coordinate": [
                47.39775097793666,
                8.5455993642297585,
                25
            ],
            "frame": 3,
            "id": 1,
            "param1": 0.26179939000000002,
            "param2": 0,
            "param3": 0,
            "param4": 0,
            "type": "missionItem"
        },
        {
            "autoContinue": true,
            "command": 16,
            "coordinate": [
                47.39802331778489,
                8.5455242657949952,
                25
            ],
            "frame": 3,
            "id": 2,
            "param1": 0,
            "param2": 0,
            "param3": 0,
            "param4": 0,
            "type": "missionItem"
        },
        {
            "autoContinue": true,
            "command": 3000,
            "coordinate": [
                47.398520784522034,
                8.5454330642202763,
                25
            ],
            "frame": 2,
            "id": 3,
            "param1": 4,
            "param2": 0,
            "param3": 0,
            "param4": 0,
            "type": "missionItem"
        },
        {
            "autoContinue": true,
            "command": 16,
            "coordinate": [
                47.399860658534699,
                8.5451272920559518,
                25
            ],
            "frame": 3,
            "id": 4,
            "param1": 0,
            "param2": 0,
            "param3": 0,
            "param4": 0,
            "type": "missionItem"
        }
    ],
    "plannedHomePosition": {
        "autoContinue": true,
        "command": 16,
        "coordinate": [
            47.397742000000001,
            8.5455939000000001,
            0.255
        ],
        "frame": 0,
        "id": 0,
        "param1": 0,
        "param2": 0,
        "param3": 0,
        "param4": 0,
        "type": "missionItem"
    },
    "version": "1.0"
}
`

DroneDude123, 2016-02-25:
[simple_vtol_mission.mission.txt](https://github.com/PX4/Firmware/files/146906/simple_vtol_mission.mission.txt)

LorenzMeier, 2016-02-26:
Fixed on master.

closed 2016-02-26 by LorenzMeier