PX4/PX4-Autopilot #3864: VTOL logs current master

sanderux, 2016-02-26:
For reference here's 2 logs for VTOL QuadRanger on current master.
For tuning back transition I flew first on:

MPC_ACC_HOR_MAX = 4
http://logs.uaventure.com/view/4Dfk6N65x2hDNeRNccQZzM#Actuators_PLOT

Then, much smoother back transition
MPC_ACC_HOR_MAX = 2
http://logs.uaventure.com/view/djVcLjJZChq3QsUmoBsrWh#Actuators_PLOT

FYI @tumbili @LorenzMeier @AndreasAntener

AndreasAntener, 2016-02-27:
Are you suggesting we should make  MPC_ACC_HOR_MAX = 2 default? If yes, please add to rc.vtol_defaults :+1:

sanderux, 2016-02-27:
Maybe you could test it on the funcub my cruise speeds are higher so it might not be applicable to all

AndreasAntener, 2016-02-28:
I was always using 2

sanderux, 2016-02-28:
When looking at your jan logs i think you were on 4.
I think this is mostly applicable to quadplanes. You think it would help for other configd?

AndreasAntener, 2016-02-28:
Ok, I was using 2 since a while. It makes absolute sense, I would even consider 1 as a default. It helps quadplanes and tilt-rotors, and it won't hurt tailsitters. So I'd suggest setting it in rc.vtol_defaults

closed 2016-02-28 by sanderux