PX4/PX4-Autopilot #3866: Startup mode

dagar, 2016-02-26:
I'd like the ability to startup into a non manual mode (fixed wing [airspeed](https://github.com/PX4/Firmware/pull/3861) mode). Is there a way to do this currently?
If not I was thinking of adding the ability to set the mode a commander argument and then adding it to the end of my airframe init script. Alternatively it could be a param.

LorenzMeier, 2016-02-26:
Already allowed non-manual on master. You can just add it. It uncovered corner cases though.

dagar, 2016-02-26:
Are you referring to arming in non manual https://github.com/PX4/Firmware/issues/3853 or something else?
I'm talking about already being in airspeed mode by the end of bootup, preventing the user from ever seeing man or stab.

LorenzMeier, 2016-02-26:
Yes, you can do that today on master

closed 2016-02-26 by dagar