PX4/PX4-Autopilot #3867: Bring travis builds in sync

julianoes, 2016-02-26:
The Mac OS build seems to have different warnings enabled. This makes travis fail for me even though it compiles fine on my system, see: https://travis-ci.org/PX4/Firmware/builds/112058279.

dagar, 2016-02-28:
This might just be different versions of clang. What do you have locally @julianoes 

travis-ci reports

    g++ --version
    Configured with: --prefix=/Applications/Xcode.app/Contents/Developer/usr --with-gxx-include-dir=/usr/include/c++/4.2.1
    Apple LLVM version 7.0.0 (clang-700.0.72)
    Target: x86_64-apple-darwin14.5.0

julianoes, 2016-02-29:
Well it also compiles fine for the travis Ubuntu build, it's just the Mac OS that is different.

Locally I have:
```
g++ --version
g++ (Ubuntu 5.2.1-22ubuntu2) 5.2.1 20151010
```

dagar, 2016-02-29:
I found the problem. Wextra is being set after Wno-sign-compare which is why it's being overridden and reenabled by clang. I'll fix it.

commit 5a167ea9 (2016-03-01): reorder compiler options to respect -Wno-sign-compare

-closes #3867
closed 2016-03-01 by LorenzMeier