PX4/PX4-Autopilot #3870: POSCTL: Reduce delay for users

LorenzMeier, 2016-02-26:
@tumbili The user feedback is that POSCTL is sluggish. I think we at least need to expose the configuration options, if not adjust it again.

LorenzMeier, 2016-02-26:
@gervaisK

GervaisK, 2016-02-27:
Thx. I would highly appreciate a preset, where POSCTL behaves like the awesome ALTCTL mode in terms of lock in, breaking, corrections and speed.

LorenzMeier, 2016-02-27:
These params should already largely do the trick:

MPC_ACC_HOR_MAX = 10
MPC_HOLD_MAX_Z = 1
MPC_HOLD_MAX_XY = 1

GervaisK, 2016-02-27:
MPC_HOLD_MAX_Z = 1
MPC_HOLD_MAX_XY = 1

I set them to 1.5. and MPC_ACC_HOR_MAX = 10 and right, that does the job to 70% , its faster, but fast movements and reverse still don´t come with an effective brake like @ ALTHOLD,its more of a soft swing. Still an great improvement. Thx.

AndreasAntener, 2016-02-28:
@LorenzMeier @tumbili to throw in an idea here: maybe velocity control isn't the right approach for stick input. As long as the sticks are moving we could just leave it in attitude control and switch to velocity control (and eventually position hold) when the sticks are inside the dead zone.

AndreasAntener, 2016-02-28:
@GervaisK I'd be interested in a log of one of your test flights with your suggested parameters. Especially I'd like to know how it reacted to your stick inputs in position control. In our experience it reacts quite aggressively to stick input changes (high initial roll rate setpoint). Which is why we use low  MPC_ACC_HOR_MAX.

LorenzMeier, 2016-02-28:
@AndreasAntener I don't think so. That would mean not offsetting wind, which is a bad user experience. Its just a matter of tuning the system to give the user the right feel.

AndreasAntener, 2016-02-28:
That's a good argument. Of course I have nothing against it if tuning resolves it.
My wording before didn't quite match what I tried to say: Instead of not being the "right" approach it might not be the only valid approach. My proposal would go into the direction what @kd0aij does with altitude control, basically having manual control unless you don't move your stick outside the dead zone, then the system takes over.

GervaisK, 2016-02-28:
@ andreas : >  basically having **manual** control 
Well, I´d personally still need baro assistance outside that dead zone.

I have to do new logs (flight) since I did meanwhile about 20 short flights with several adjustments in order to get a bigger Althold dead band, because its still a bit tricky to stick at a defined height, so  

What do you need ? How long (up to 8 minutes) ?

AndreasAntener, 2016-02-29:
Any flight will do, 30sec flying in posctl going left and right for example.

GervaisK, 2016-02-29:
Its in your uaventure mailbox .

closed 2016-03-13 by LorenzMeier