PX4/PX4-Autopilot #3871: VTOL back transition mission behavior

sanderux, 2016-02-26:
When performing a back transition from a mission the MC tries to return to the path.
For a smoother transition it would be better to ignore the deviation and continue towards the next waypoint.

[image]

LorenzMeier, 2016-02-26:
I think that's a non-issue. You are just looking at the aesthetics of the line, but what counts is if the vehicle did stick to the mission. You probably will want to reduce the L1 radius / length for the SITL VTOL, which will make it comply far better with the mission.

closed 2016-02-26 by LorenzMeier

sanderux, 2016-02-26:
I would add that this also happens in larger real missions: http://logs.uaventure.com/view/djVcLjJZChq3QsUmoBsrWh

LorenzMeier, 2016-02-26:
It doesn't really matter. The mission logic is to track the lines between waypoints. Taking shortcuts is dangerous and unexpected.

sanderux, 2016-02-26:
So is the FW plotting a path to start loitering? it seems to start deviating right before transition. If this is the case it could consider the upcoming transition.