PX4/PX4-Autopilot #3872: VTOL: Back transition timing.

tubeme, 2016-02-27:
Hi Guys,

Probably it is a strange request but we've had numerous dire situations flying in FW flight where we had stalled or there is some uncontrolled fall etc. We were thinking if there should be a way to recover from this situations with back transition to quad. But there is a problem, we have preset back transition time to 3 sec. in order to be smooth. In most emergency situations these 3 seconds are fatal.

So if we could trigger immediate mandatory back transition in the Taranis HS switch? We use HS switch for parachute launch usually. Can be connected to AUX3.

Then when in control, we can  just switch back to quad from the main dial and again to FW for another transition.

LorenzMeier, 2016-02-27:
Closing this in favor of https://github.com/PX4/Firmware/issues/3823.

@tubeme If you regularly stall in fixed wing you have set up your plane parameters incorrectly. The stall speed that you have provided is awfully low (our experience suggests a stall speed of 11-14 m/s).

Given that you have a lot of stall related issues, have you considered that your assumptions about the plane might be fundamentally wrong and that your very unstable results are mostly driven by that? We almost never stall, in particular not in forward flight.

closed 2016-02-27 by LorenzMeier

tubeme, 2016-02-27:
During our experiments, sometimes we are flying on the edge on purpose, to see where it can go. So we try everything in its extremes. So in this situations sometimes inevitable happens. Of course we will not use this settings for release version. The frame fly very good I assure you and it is very economical.