PX4/PX4-Autopilot #3874: Mission: DO_JUMP sends MISSION_ITEM to ground station

DonLakeFlyer, 2016-02-27:
* QGC + SITL
* Create mission with an infinite loop do jump
* Start mission

When the vehicle hits the DO_JUMP it correctly justs but the firmware sends a MISSION_ITEM message to QGC which cause QGC to pop an error since you aren't supposed to get one of these for no reason.

LorenzMeier, 2016-02-28:
The intent was to allow the GCS to track how many do jumps are remaining. Is the spec wrong or the use here?

DonLakeFlyer, 2016-02-28:
"This message is emitted to announce the presence of a mission item and to set a mission item on the system."

Hard to say. QGC never expects to get a MISSION_ITEM outside of a waypoint protocol sequence. Not sure if updating that directly in the mission item is a good thing. What happens in this case:
* Send mission with jump count to vehicle
* Start mission
* Let it do some jumps
* Read the mission back from vehicle
Now the jump command count is different than when you sent it the first time?

LorenzMeier, 2016-02-28:
Yes, that's the current implementation. The count goes down (for one having a count) until it reaches zero. Just wondering if that makes sense or not.

DonLakeFlyer, 2016-02-28:
I guess it's sort of interesting to get remaining count. But if so, I think I'd do it on another message. I'd also only change the count on a side variable internally so you can read back to mission and get back exactly what you sent.

dagar, 2017-04-11:
Was this resolved?

dagar, 2017-04-18:
@DonLakeFlyer is this still an issue?

DonLakeFlyer, 2017-04-18:
If nothing has changed on the firmware side I would assume so.

dagar, 2017-04-28:
Are you interested in this @sanderux?

sanderux, 2017-04-28:
Yes, but maybe others can pitch in on the method as i am unclear on that

dagar, 2017-04-28:
So it's resending the mission item each time with the count, but it sounds like you and @DonLakeFlyer want an explicit mavlink message?

sanderux, 2017-04-28:
I think it should be a field here: http://mavlink.org/messages/common#MISSION_CURRENT

closed 2018-02-21 by PX4BuildBot