PX4/PX4-Autopilot #3877: Low battery results in arming lockdown, but user feedback is confusing

LorenzMeier, 2016-02-28:
```
WARN  offboard mission updated: dataman_id=0, count=3, current_seq=0
INFO  TONE_SET_ALARM 4
INFO  TONE_SET_ALARM 7
telem> LOW BATTERY, LOCKING ARMING DOWN
INFO  TONE_SET_ALARM 8
telem> Preflight check resolved, reboot to complete
INFO  TONE_SET_ALARM 5
INFO  TONE_SET_ALARM 5
INFO  TONE_SET_ALARM 8
telem> Preflight check resolved, reboot before arming
INFO  TONE_SET_ALARM 5
INFO  TONE_SET_ALARM 5
INFO  TONE_SET_ALARM 8
```

The preflight check resolved messages have to go.

LorenzMeier, 2016-04-01:
@kd0aij Would you have time to look into src/modules/commander/state_machine_helper.cpp and add the missing user output?

kd0aij, 2016-04-01:
https://github.com/PX4/Firmware/pull/4140

LorenzMeier, 2016-04-18:
Fixed this.

closed 2016-04-18 by LorenzMeier