PX4/PX4-Autopilot #3879: Add min/max/increment/unit/decimal meta data to main parameters

LorenzMeier, 2016-02-28:
@kd0aij This will fix the Android UI for you: We need to add meta data like in this commit to all major params:
https://github.com/PX4/Firmware/commit/e3853ed4e536860cc3b2f4e688f0c20eb2e2151b

```
@unit m/s
@min 0.0
@max 100.0
@increment 0.5
@decimal 2
```

LorenzMeier, 2016-02-28:
@sanderux Can you help with that for VTOL params?

kd0aij, 2016-02-28:
@LorenzMeier Thanks. Will there be additional buttons for increment/decrement of each individual parameter, or will there be a new screen for tuning groups of parameters?

LorenzMeier, 2016-02-28:
It will just make using the tablet easier with a better value selector, which will apply to existing and potential new tuning screens.

closed 2016-03-24 by LorenzMeier