PX4/PX4-Autopilot #3886: mixer output limits are partially broken

kd0aij, 2016-02-29:
appears to be a parsing problem with output limits: 
M: 2
O:        7500    7500    0  -7500   7500
S: 1 0  -10000  -10000    0 -10000  10000
S: 1 1   10000   10000    0 -10000  10000

M: 2
O:        7500    7500    0  -7500   7500
S: 1 0  -10000  -10000    0 -10000  10000
S: 1 1  -10000  -10000    0 -10000  10000

results in the second channel being dead when the mixer is on SD card, but works properly with mixer in ROMFS. Change to:

M: 2
O:        7500    7500    0  -7500  10000
S: 1 0  -10000  -10000    0 -10000  10000
S: 1 1   10000   10000    0 -10000  10000

M: 2
O:        7500    7500    0  -7500   7500
S: 1 0  -10000  -10000    0 -10000  10000
S: 1 1  -10000  -10000    0 -10000  10000

and the second channel is also dead with the mixer in ROMFS.
The above lines are changes to the mixer in https://github.com/PX4/Firmware/pull/3875

LorenzMeier, 2016-03-24:
@kd0aij Can you test again on master? The reason I ask is that tridge fixed a buffer overflow which might explain why it depends on individual numeric values.

tumbili, 2016-04-21:
@kd0aij Did you get a chance to test this?

kd0aij, 2016-04-22:
I'll have to dust off my tailsitter

LorenzMeier, 2016-04-22:
Moving this to 1.4.0 since every body else is happy.

LorenzMeier, 2016-05-22:
Outdated / most likely fixed.

closed 2016-05-22 by LorenzMeier