PX4/PX4-Autopilot #3889: Broken PWM Outputs for VTOL Master

tumbili, 2016-02-29:
When using master on my 13001 Caipiroshka my ESCs do not receive the disarmed signal and nothing happens on arming.
"pwm info" gives:
```
device: /dev/pwm_output0
channel 1: 0 us (alternative rate: 400 Hz failsafe: 0, disarmed: 900 us, min: 1080 us, max: 2000 us)
channel 2: 0 us (alternative rate: 400 Hz failsafe: 0, disarmed: 900 us, min: 1080 us, max: 2000 us)
channel 3: 0 us (alternative rate: 400 Hz failsafe: 0, disarmed: 900 us, min: 1000 us, max: 2000 us)
channel 4: 0 us (alternative rate: 400 Hz failsafe: 0, disarmed: 900 us, min: 1000 us, max: 2000 us)
channel 5: 0 us (default rate: 50 Hz failsafe: 0, disarmed: 0 us, min: 1000 us, max: 2000 us)
channel 6: 0 us (default rate: 50 Hz failsafe: 0, disarmed: 0 us, min: 1000 us, max: 2000 us)
channel group 0: channels 1 2 3 4
channel group 1: channels 5 6

```

Obviously the pwm values should be 900 us and not 0 us.

tumbili, 2016-02-29:
@kd0aij I was told to point the gun at you ;-)

kd0aij, 2016-02-29:
fmuv1,2 or 4? At one time the kill switch was turning PWM off, but that was in "lockdown" mode.

tumbili, 2016-02-29:
fmuv4, so Pixracer

tumbili, 2016-02-29:
@kd0aij

kd0aij, 2016-02-29:
sounds like your config is broken too though :-)
My VTOL is fmuv2

tumbili, 2016-02-29:
Hm, any idea where to look?

kd0aij, 2016-02-29:
do you have the kill switch mapped?

kd0aij, 2016-02-29:
and your problem is similar to the one I had with channel 6 being dead, but that seemed to be a mixer textfile parsing problem.

tumbili, 2016-02-29:
Let me check

tumbili, 2016-02-29:
No, kill switch is not mapped

kd0aij, 2016-02-29:
Why would arming work on a PixRacer quad, but not on VTOL? There should be no difference, should there?

kd0aij, 2016-02-29:
There is a circuit breaker to disable the safety switch, if you don't want to use one.

kd0aij, 2016-02-29:
This was the PR for the "painless" addition of safety switch handling: https://github.com/PX4/Firmware/pull/3827
I tested it on my PixRacer quad, and it worked there after a few false starts.

tumbili, 2016-03-01:
Fixed, on master.

closed 2016-03-01 by tumbili