PX4/PX4-Autopilot #3890: PROPOSAL: better handling of MAVLINK RC messages

scott-eddy, 2016-02-29:
I'd like to expand the discussion of #3309 to include cases for handling RC_CHANNELS_OVERRIDE.  While I'm aware of the shortcomings of controlling the vehicle via rc override values there are still some cases in which the control scheme makes sense, and if nothing else the flight stack should have a proper implementation of handling that message for those who want to use it.  

I propose the following changes to those mentioned in #3309:
- The proposed joystick_input.msg file be expanded to mavlink_rc_input.msg which is capable of storing values coming from both RC_CHANNELS_OVERRIDE, and MANUAL_CONTROL.  This message should also store the MAVLINK source of the data similar to how is done with input_rc.msg.
- Whereas the joystick data is locked out if the RC control state changes and is in manual, the OVERRIDE data is not locked out and instead the RC control is regardless of state.
- input_rc.msg be altered to such that RC_INPUT_SOURCE no longer contains MAVLINK as any mavlink sources are now handled separately.

The only case I can imagine being a problem is when a user wants to use both RC_CHANNELS_OVERRIDE and a joystick.  But I think that it's such a low priority use-case it can be handled after this is implemented.  

I'd be happy to start work on this right away if it would be merged in.  @LorenzMeier what do you think about the modifications?

DonLakeFlyer, 2016-02-29:
I don't quite get what you are trying to accomplish with this? Some sort of automatic hand-off between Radio and Override? RC_OVERRIDE alread has manual handoff mechanism built into it for it to take over and/or hand back to radio control.

scott-eddy, 2016-02-29:
RC_OVERRIDE does have a manual hand-off mechanism, but once the message gets parsed by the [mavlink receiver](https://github.com/PX4/Firmware/blob/ebeb9eba4bc8255f07b8bdba485638d3694cccdf/src/modules/mavlink/mavlink_receiver.cpp#L1095) the firmware doesn't handle that hand-off mechanism.  This is because once the mavlink messages are parsed by the receiver and published to input_rc the sensors application never checks to see what actual link generated the channel values before publishing them to [rc_channels](https://github.com/PX4/Firmware/blob/ebeb9eba4bc8255f07b8bdba485638d3694cccdf/src/modules/sensors/sensors.cpp#L1954).

What I'm trying to accomplish with this is:
1. Differentiating what ORB_ID the input rc values are being published to based on their physical link (e.g. MAVLINK vs. Controller)
2. Allow for handling which RC values take precedence in the controllers given their source (both physical and MAVLINK message type) and the vehicle's state.

Does that make more sense?

scott-eddy, 2016-03-02:
@AndreasAntener What do you think about all this?

LorenzMeier, 2016-05-22:
@scott-eddy This interested in this?

LorenzMeier, 2017-04-30:
@scott-eddy We've improved this on master - did this solve your issue or do you still need additional fixes?

closed 2018-02-21 by PX4BuildBot