PX4/PX4-Autopilot #3893: FW: handle loss of and/or not having an airspeed sensor

sanderux, 2016-02-29:
As discussed in dev call 2016-02-29
In reference to https://github.com/PX4/Firmware/issues/3834
And for VTOL expanding on https://github.com/PX4/Firmware/issues/3850

FW flight currently depends on an airspeed sensor in auto mode, and for VTOL it is even required for transition. Given a properly tuned config (cruise throttle with maybe some pitch mixing) it should be possible to fly without an airspeed sensor. This flight mode could be desired for small airplanes that do not have an airspeed sensor, but it would also add the ability to failsafe the loss of reliable airspeed readings during flight.

I propose to implement falling back to cruise throttle and mixing in pitch angle when either an airspeed sensor is missing, or the reading becomes unreliable. In the latter case I would propose a RTL failsafe event.

@tumbili @AndreasAntener your thoughts please

AndreasAntener, 2016-03-01:
I consider flying at a fixed throttle (higher than cruise, because cruise is for efficient flight) a valid option as airspeed failsafe.

sanderux, 2016-07-23:
implemented

closed 2016-07-23 by sanderux