PX4/PX4-Autopilot #3894: MPC Final approach land speed

sanderux, 2016-02-29:
Implement parameter to set final approach altitude and land speed.
This would allow slower descend rate for the final x meters.

@LorenzMeier I propose this with a default value of 1.5m above home and around 60% land speed. Any thoughts?

ecmnet, 2016-02-29:
What about setting the landing speed dynamically based on the altitude? I had some good results by setting the speed to 0.5 * altitude beginning with 2m and let it drop some cm above ground.

sanderux, 2016-02-29:
I think that would only provide good results with sonar or lidar. without that you would risk touching the ground and tipping over as baro/gps altitude is not that acurate

ecmnet, 2016-02-29:
True, I had lidar running.

AndreasAntener, 2016-02-29:
@sanderux what you just said applies in general for this. A slow landing speed by default is good enough IMHO. A 2 phased landing approach should either be implemented with a range finder or if you fly a mission descend to a lower altitude already before the actual landing.

sanderux, 2016-02-29:
@AndreasAntener Of course you could set a lower altitude before initiating the land. but in practice people (like me) are lazy. To land a VTOL for instance it already requires 3 mission items; Last waypoint before transition, transition and then land. Adding a forth just to decrease the land speed seems a bit much.

Most airframes can easily descend  2~3 m/s but that would give quite a big impact on landing. A RTL event by efault would even initiate a land at 60m altitude, at the current setting it would take 2 full minutes to land. If we add this parameter i think a lot of people will use that to set a higher default landing speed and safe the trouble of having to set a lot of mission items.

did i convince you? :)

sanderux, 2016-02-29:
As for the inaccurate altitude estimation, the final approach altitude could default to 3 or 4 meters above the landing site.

AndreasAntener, 2016-02-29:
No :) I'm not arguing about the feature, I'd like to have it. But it's unsafe to assume ground height when you actually don't know it. The landing point could be anywhere. Even with a terrain model I wouldn't do it. The only option for me here is to do that with a range finder and I would gladly support you in adding this.

sanderux, 2016-02-29:
Granted, it's not ideal when start and end point are not on the same altitude. however; i think 9%% of the time we land on or near the same location (or at least the same altitude). Anyone who does missions landing in different locations should indeed use a range finder and we could note this when changing the land speed and/or final approach params.

The code would not assume anything, it would simply lower descent rate at a given altitude above home altitude. It doesn't change anything to the current settings, only those that specifically set this would benefit and they probably know (or have been told) what it means.

thnks for your support, i'll get to work on it :)

dagar, 2016-11-08:
What's the conclusion here? I'd like to optionally be able to enable this.

dagar, 2017-12-06:
Implemented.

closed 2017-12-06 by dagar