PX4/PX4-Autopilot #3899: FW failsafes during TAKEOFF/LANDING

dagar, 2016-03-01:
I had a brief data link loss causing a switch to RTL during a fixed wing auto takeoff yesterday. It was quick and resumed the mission, but it could have caused a crash. Would it make sense to limit this failsafe when flying a SETPOINT_TYPE_TAKEOFF or SETPOINT_TYPE_LAND? At the very least during climbout for takeoff and flare for landing.

LorenzMeier, 2016-03-01:
Not sure. It would have climbed like on a takeoff and loitered. That seems safer than any other approach?

dagar, 2016-03-01:
Early in the takeoff (altitude_error > _parameters.climbout_diff) roll is constrained and I'm looking at also limiting pitch for larger planes before they reach min airspeed. With home directly behind the plane and in RTL wouldn't it just be a regular position without any special constraints? It's definitely an edge case, but maybe the right solution is adding in_takeoff_situation() logic to SETPOINT_TYPE_POSITION?

closed 2016-09-04 by dagar