PX4/PX4-Autopilot #3902: FW auto takeoff issues

dagar, 2016-03-01:
Yesterday I attempted a fixed wing auto takeoff with the takeoff waypoint ~50-60m to the south at 25m above home. I briefly switched into AUTO:MISSION to verify the throttle, then to MANUAL, and then back into AUTO for the throw. It appears that the takeoff waypoint was immediately accepted and that the actual throw was in a regular waypoint. I haven't been able to fully piece together the details, but I think this could make sense with the actual acceptance radius calculated in the l1 controller.

[image]

This failure is arguably user error, but could it be improved? What about these options?
 - tighten takeoff waypoint acceptance criteria
    - require !landed, maybe min airspeed for FW
 - don't allow FW to enter AUTO if landed and the current waypoint is not takeoff

tumbili, 2016-03-01:
Log: http://logs.uaventure.com/view/zEawE6UvMjTaAiHU8REtzi

dagar, 2016-03-01:
If you think there's something to fix or improve here I can spend some time implementing and testing.

dagar, 2016-03-01:
I'm failing to see how the l1 distance could explain the immediate acceptance of the takeoff waypoint

    _L1_distance = _L1_ratio * ground_speed
             = (1.0f / M_PI_F * _L1_damping * _L1_period) * ground_speed
             ~= (1 / 3.14159 * 0.75 * 20) * 1
= **4.77**

Am I missing something here?

dagar, 2016-03-01:
My understanding of navigator, missions, and l1_control is currently pretty basic, but I'm wondering if this could be a race condition with l1_control not properly initialized?

 1. switch to AUTO for the first time
 2. navigator initially publishes empty pos_sp_triplet with current.valid=false
 3. fw_pos_ctrl control_position() returns true because current.valid=false and doesn't run anything that calls into l1 - https://github.com/PX4/Firmware/blob/master/src/modules/fw_pos_control_l1/fw_pos_control_l1_main.cpp#L2148
     - turn_distance = _l1_control.switch_distance(100.0f); **<-- l1 unitialized at this point**
     - navigation_capabilities_publish()
 4. navigator runs and through Navigator::get_acceptance_radius() uses some garbage value from uninitialized _L1_distance
 5. takeoff waypoint is immediately accepted

dagar, 2016-03-01:
I'm attempting to reproduce in HIL with a debugger.

dagar, 2016-03-03:
I've been unable to reproduce this failure in HIL using the same waypoints and starting location. Even if the underlying problem is still present the fix to only accept waypoints if not landed should address it. Closing.

closed 2016-03-03 by dagar