PX4/PX4-Autopilot #3905: Pixracer mag cal fails on current master

sanderux, 2016-03-01:
it does work on stable

```
[cal] up orientation detected
[cal] Rotate vehicle around the detected orientati
[cal] Continue rotation for 7 seconds
[cal] up side done, rotate to a different side
[cal] up side done, rotate to a different side
[cal] ERROR: Replace board
ERROR: Replace board, fault in mag 0
Excessive offsets:   0.0532,  -0.5191,  -0.4636
[cal] calibration failed: mag
```

LorenzMeier, 2016-03-02:
Those offsets are almost hitting the absolute maximum values. Are you sure this board is fine? I widened them further on master, but in general an offset like that is indication of something not being quite right. The last case we had like that was a mag with bad solder connection.

LorenzMeier, 2016-03-02:
Closing this. The error message is real. If it still doesn't work, replace your board.

LorenzMeier, 2016-03-02:
@pkocmoud @nickarsov This is the 2nd Pixracer with this issue. The first was "fixed" with some hot air onto the mag.

closed 2016-03-02 by LorenzMeier

sanderux, 2016-03-02:
[image]

kd0aij, 2016-03-02:
@LorenzMeier What could be wrong with PixRacer to cause the external mag cal problem?

LorenzMeier, 2016-03-02:
There was never an issue with the external mag. The indications were all correct.