PX4/PX4-Autopilot #3911: Mag0 Calibration Fail Quanum LEA6H GPS+Compass

tubeme, 2016-03-03:
Hello Lorenz,

I have problems calibrating the external Mag0.
I changed the Pix board with AUAV-X2 brand new. Calibrated the Airspeed to check if I2C works so it is OK.
Here I made 2 calibrations:

ERROR: Replace GPS unit, external mag fault
Excessive offsets:   1.2199,   0.0402,  -0.3065, #

ERROR: Replace GPS unit, external mag fault
Excessive offsets:   1.1905,   0.0473,  -0.2969, #

And then changed the Quanum GPS+Compass with a brand new one. Even worse results:

ERROR: Replace GPS unit, external mag fault
Excessive offsets:   1.4110,   0.1219,  -0.3032, #

It gave me a vibration warning a couple of times...

HIGH VIBRATION! g: 14 a: 63 m: 9

Cannot calibrate the Mag0.

Any ideas?

LorenzMeier, 2016-03-03:
Are you sure you don't have a huge piece of metal close to your mag? Or a motor coil? Those offsets are not ok to fly with.

tubeme, 2016-03-03:
I'm sure there is not metal nor motors around the mag. It is placed at the wing in a special place made for it with fully extended cable far away from everything that might interfere. Also we were flying all this hardware before with older version of PX4, We have flown it in two Quads and since last jan we have flown it in our VTOL, including the broken HKPilot32. From the calibration offset of 1.2 and 1.4 for one of the axis es with a scale of 1.4 gauss it seems that it is fully rotated around this axis. There is something very wrong with the calibration in my case... the one with 1.4 is a brand new unit with a 3 flights on it. Never crashed nor had a damage. Also the AUAV-X2 is brand new, never flown.

sanderux, 2016-03-03:
what happens when you disconnect the external gps/mag and run compass calib?

tubeme, 2016-03-03:
When only the internal compass is working, ie I disconnected the external compass, the calibration goes OK.

Then again when connected the calibration does not go well. I get High Vibration warning between 1-2 step of cal almost all the time. The calibration offset for the brand new GPS  is consistent every time:

ERROR: Replace GPS unit, external mag fault
Excessive offsets:   1.4182,   0.1073,  -0.3079, #

Is it possible that all the GPS units we buy are defective? with so much offset. We buy probably the most expensive one from HK. We've never had any problems with them until 1 month ago, when we flashed latest master at that time. We had a very successful flights in a VTOL config with this offsets:

CAL_MAG0_XOFF	1.1486901
CAL_MAG0_XSCALE	0.99373263
CAL_MAG0_YOFF	0.038466614
CAL_MAG0_YSCALE	1.0996851
CAL_MAG0_ZOFF	-0.2766014
CAL_MAG0_ZSCALE	1.0077993

and this PX4 version:

Firmware Version	0fed23a0006854ff4936bce2f1b19b3862ccad39

So at that time it is obvious that PX4 accepted this offsets and flew well with that.

I can upload a Log if you need to prove it.

Or the other thing could be some specifics to the Quanum Lea6H and its mag.

tubeme, 2016-03-03:
I will disconnect the external Mag for now because we have to fly...

sanderux, 2016-03-03:
@tubeme you say you have extended your gps lead. could you try it with the default cable length?

LorenzMeier, 2016-03-03:
Those offsets are way too high. They are not safe to fly because you won't get a correct heading estimate. It doesn't matter if flying worked for you before, this is not safe and the system is rejecting to accept the calibration for very good reasons.

sanderux, 2016-03-03:
@tubeme also try to connect ONLY the mag (not the GPS) and see how that does
just to narrow down where the issue comes from. As Lorenz stated, it's not safe to fly on these values.

tubeme, 2016-03-03:
@sanderux I did not extend the cable. I use the original cable but fully extended. I cannot connect the ext mag only because it gets its power from the GPS serial. So when I power the gps the mag is powered as well. I have only data cables grom the ext mag go in the I2C splitter.
@LorenzMeier OK I understand I will buy a new one. Can you suggest GPS - supplier, kind I saw there was UAVCAN GPS+Mag? I wish I had this result earlier in order to return the GPS's... 

Thank you both for the reply.

mhkabir, 2016-03-04:
If you're looking to get a CAN GNSS, you should have a look at Zubax GNSS v2. I'm not sure if it's retailing yet, but the prototype I have here outperforms any other GPS system I've used. 
I highly recommend getting CAN hardware to replace i2c ones. Especially when reliability is a factor.

kd0aij, 2016-03-04:
Why does the Zubax gps perform so well? Larger antenna?

mhkabir, 2016-03-04:
For starters, the antenna is slightly larger, yes. It also has inbuilt power filtering, which I've seen is effective against noisy onboard hardware like lidars. The RF backend also seems to be well built. 
Comparing side by side, I've had a ublox m8n and a Zubax module indoors, and Zubax locked onto 16 sats, the m8n 11. Quite an improvement :)

@pavel-kirienko should be able to give you the technical details.

closed 2016-03-11 by tubeme