PX4/PX4-Autopilot #3913: Standard AETR plane can not be used

zengdechi, 2016-03-03:
I selected Standard AETR plane in airframe column,but it can not be used.  prompting "warning:system boot incomplete,check config" and "critical:IO vscale upload failed".And radio can not be calibrate.

tubeme, 2016-03-03:
Can you upload your boot log?

zengdechi, 2016-03-03:
how to upload boot log?

tubeme, 2016-03-05:
Get the SD card from the Pixhawk. Inside the card find the files:

bootlog.txt and one from the flight tha looks like this:

msgs_2016_03_04_15_37_32.txt

You have to find the one from your flight and upload it here like a zip file.

zengdechi, 2016-03-05:
thank you! I will try.

darthrake, 2016-03-05:
I have probably the same problem with AERT. Here is a boot log from PX4 stable, clean SD card, just setup as AERT.

```
sercon: Registering CDC/ACM serial driver
sercon: Successfully registered the CDC/ACM serial driver
[i] microSD mounted: /fs/microsd
param: selected parameter default file /fs/mtd_params
[param] Loaded: /fs/mtd_params
rgbled on I2C bus 2 at 0x55 (bus: 100 KHz, max: 100 KHz)
 1: + SYS_AUTOCONFIG: match
 1:   SYS_USE_IO: match
 2101: + SYS_AUTOSTART: match
param: Error: Parameter NAV_LAND_ALT not found.
param: Error: Parameter NAV_RTL_ALT not found.
param: Error: Parameter NAV_RTL_LAND_T not found.
  NAV_ACC_RAD: curr: 10.00000 -> new: 50.00000
  PE_VELD_NOISE: curr: 0.3000 -> new: 0.3500
  PE_POSD_NOISE: curr: 1.2500 -> new: 1.0000
+ SYS_AUTOCONFIG: curr: 1 -> new: 0
px4io: CRCs match
dataman: Unknown restart, data manager file '/fs/microsd/dataman' size is 103090 bytes
MS5611_SPI on SPI bus 1 at 3 (20000 KHz)
bst: no devices found
adc init done
ver hwcmp match: PX4FMU_V2
HMC5883_I2C on I2C bus 1 at 0x1e (bus: 100 KHz, max: 400 KHz)
hmc5883: no device on bus 1
mpu6000: driver start failed
MPU6000 on SPI bus 1 at 4 (1000 KHz)
L3GD20 on SPI bus 1 at 1 (11000 KHz)
LSM303D on SPI bus 1 at 2 (11000 KHz)
meas_airspeed: no MS4525 airspeed sensor connected
ets_airspeed: no ETS airspeed sensor connected
ets_airspeed: no ETS airspeed sensor connected
nsh: sf10a: command not found
px4io default PWM output device
mavlink_if0: mode: 0, data rate: 1200 B/s on /dev/ttyS1 @ 57600B
mavlink_if0: offboard mission init: ERROR
 157600:   SYS_COMPANION: match
mavlink_if1: mode: 3, data rate: 1000 B/s on /dev/ttyS2 @ 57600B
sdlog2: log buffer size: 12288 bytes
[i] FIXED WING
gps: ubx checksum err
gps: module found: UBX
mavlink_if1: Disabling hardware flow control
commander_low_prio: settings saved.
mixer: fmu sent: "00 10000 0 -10000 10000
"
px4io mixer send error -1
mixer: error loading mixers from /etc/mixers/AERT.main.mix
mixer: failed to load mixer
[i] Error loading mixer: /etc/mixers/AERT.main.mix
pwm: reading disarmed value of zero, disabling disarmed PWM
pwm: failed get disarmed values

NuttShell (NSH)
nsh>
```

darthrake, 2016-03-05:
with `debuglevel 6`:

```
[i] FIXED WING
mixer: fmu sent: "M: 1
O: 10000 10000 0 -10000 10000
S: 0 0 10000 10000 0 -10"
mixer: IO CONSOLE: ՙ����59

mixer: fmu sent: "000 10000
M: 1
O: 10000 10000 0 -10000 10000
S: 0 1 -10000 "
mixer: IO CONSOLE: used 69

mixer: fmu sent: "-10000 0 -10000 10000
M: 1
O: 10000 10000 0 -10000 10000
S:"
mixer: IO CONSOLE: sed 71

mixer: fmu sent: " 0 2 10000 10000 0 -10000 10000
M: 1
O: 10000 10000 0 -1000"
mixer: IO CONSOLE: ��69

mixer: fmu sent: "0 10000
S: 0 3 0 20000 -10000 -10000 10000
M: 1
O: 10000 10"
mixer: fmu sent: "000 0 -10000 10000
S: 2 0 10000 10000 0 -10000 10000
M: 1
O"
mixer: fmu sent: ": 10000 10000 0 -10000 10000
S: 2 1 10000 10000 0 -10000 10"
mixer: fmu sent: "000
M: 1
O: 10000 10000 0 -10000 10000
S: 2 2 10000 10000 0"
mixer: fmu sent: " -10000 10000
M: 1
O: 10000 10000 0 -10000 10000
S: 2 3 100"
mavlink_if1: Disabling hardware flow control
mixer: fmu sent: "00 10000 0 -10000 10000
"
px4io mixer send error -1
mixer: error loading mixers from /etc/mixers/AERT.main.mix
mixer: failed to load mixer
[i] Error loading mixer: /etc/mixers/AERT.main.mix
```

zengdechi, 2016-03-06:
my bootlog has a error.ERROR: Could not load mixer: /etc/mixers/AETR.main.mix

SimonWilks, 2016-03-06:
It could be one of three things: firmware version, memory on the IO is
exhausted (but I'd thought this was fixed a while ago) or the IO firmware
has not updated.

Firmware: are you sure you are using the latest firmware? Either stable or
master.

IO firmware: hold the safety switch down and keep it there while applying
power to the Pixhawk. The there will be a rapid blinking of one of the red
LEDs while the IO firmware flashes. Once it's finished check if the problem
has been resolved.

Memory: if the above didn't work try selecting a different aircraft. Take
the FX79 as it has a small mixer file.

Regards
Simon

darthrake, 2016-03-06:
I'm currently using the beta branch (commit 737519), IO Firmware is up to date.

Memory: Easystar seems to work. Removing the gimbal/aux mixers works as well (AERT). The output of `px4io status` looks corrupt to me.

```
nsh> px4io status
px4io: loaded
protocol 2147483648 hardware 2147483648 bootloader 2147483648 buffer 2147483648B crc 0x8000000080000000
2147483648 controls 2147483648 actuators 2147483648 R/C inputs 2147483648 analog inputs 2147483648 relays
2147483648 bytes free
status 0x0000 SAFETY_SAFE RC_FAIL FMU_FAIL MIXER_FAIL ARM_NO_SYNC INIT_FAIL
alarms 0x0000
vservo 2147483648 mV vservo scale 2147483648
vrssi 2147483648
actuators 0 0 0 0 0 0 0 0
servos 2147483648 2147483648 2147483648 2147483648 2147483648 2147483648 2147483648 2147483648
reversed outputs: [________] trims: r:   0.0000 p:   0.0000 y:   0.0000
0 raw R/C inputs
R/C flags: 0x0000
mapped R/C inputs 0x0000
ADC inputs
features 0x0000
arming 0x0000 FMU_DISARMED IO_ARM_DENIED
rates 0x80000000 default 2147483648 alt 2147483648
debuglevel 2147483648
controls 0: 0 0 0 0 0 0 0 0
controls 1: 0 0 0 0 0 0 0 0
controls 2: 0 0 0 0 0 0 0 0
controls 3: 0 0 0 0 0 0 0 0
input 0 min 2147483648 center 2147483648 max 2147483648 deadzone 2147483648 assigned 2147483648 options 0x0000
input 1 min 2147483648 center 2147483648 max 2147483648 deadzone 2147483648 assigned 2147483648 options 0x0000
input 2 min 2147483648 center 2147483648 max 2147483648 deadzone 2147483648 assigned 2147483648 options 0x0000
input 3 min 2147483648 center 2147483648 max 2147483648 deadzone 2147483648 assigned 2147483648 options 0x0000
input 4 min 2147483648 center 2147483648 max 2147483648 deadzone 2147483648 assigned 2147483648 options 0x0000
input 5 min 2147483648 center 2147483648 max 2147483648 deadzone 2147483648 assigned 2147483648 options 0x0000
input 6 min 2147483648 center 2147483648 max 2147483648 deadzone 2147483648 assigned 2147483648 options 0x0000
input 7 min 2147483648 center 2147483648 max 2147483648 deadzone 2147483648 assigned 2147483648 options 0x0000
input 8 min 2147483648 center 2147483648 max 2147483648 deadzone 2147483648 assigned 2147483648 options 0x0000
input 9 min 2147483648 center 2147483648 max 2147483648 deadzone 2147483648 assigned 2147483648 options 0x0000
input 10 min 2147483648 center 2147483648 max 2147483648 deadzone 2147483648 assigned 2147483648 options 0x0000
input 11 min 2147483648 center 2147483648 max 2147483648 deadzone 2147483648 assigned 2147483648 options 0x0000
input 12 min 2147483648 center 2147483648 max 2147483648 deadzone 2147483648 assigned 2147483648 options 0x0000
input 13 min 2147483648 center 2147483648 max 2147483648 deadzone 2147483648 assigned 2147483648 options 0x0000
input 14 min 2147483648 center 2147483648 max 2147483648 deadzone 2147483648 assigned 2147483648 options 0x0000
input 15 min 2147483648 center 2147483648 max 2147483648 deadzone 2147483648 assigned 2147483648 options 0x0000
input 16 min 2147483648 center 2147483648 max 2147483648 deadzone 2147483648 assigned 2147483648 options 0x0000
input 17 min 2147483648 center 2147483648 max 2147483648 deadzone 2147483648 assigned 2147483648 options 0x0000
failsafe 2147483648 2147483648 2147483648 2147483648 2147483648 2147483648 2147483648 2147483648
disarmed values 2147483648 2147483648 2147483648 2147483648 2147483648 2147483648 2147483648 2147483648
```

SimonWilks, 2016-03-06:
OK, it's certainly the memory issue then.

Could you please try stable and then master if that doesn't work? I'm not sure what state beta is at, it's most likely behind stable and certainly well behind master.

darthrake, 2016-03-06:
master (8a91ce) has exactly the same issue. Status of `px4io status` looks the same.

darthrake, 2016-03-06:
Just for the sake of completeness: it does not work on stable (f5efe0) either. :airplane: :fire:

LorenzMeier, 2016-03-13:
This should be fixed on master.

closed 2016-03-13 by LorenzMeier