PX4/PX4-Autopilot #3925: VTOL: Delta Wing MAIN OUT Only Setup Problem

tubeme, 2016-03-05:
Hi, I just found a corner case while trying to rewire our VTOL sich that we use only the MAIN OUT bus and not the AUX.

So If I want to use the ch1,2 for Elevons, ch3 for throttle and then ch5,6,7,8 for the QUAD it is not possible. I want to use it because I want to give different rate for the servos. Thus setting the ch1,2 with lower rate and the rest higher rate for the motors.

So the parameter VT_MOT_COUNT 4 in my case disables the first 4 MAIN OUTs likely connected to the 4 Quad motors. In my case it just disabled the pusher motor and pushed the servos to min us the moment we reached transition air speed. 

So I had to reorder the mixer such that the first 4 channels ch1,2,3,4 are the Quad and the rest are the Elevons + pusher motor. My servos are digital so no problems with the 400hz so far. But with other servos it might be a problem.

I think I saw a way to tell PX4 in the params which motors to stop after the transition. Think it was for FireFly6... It is not a bad idea to have it enabled for the other VTOL frames as well.

LorenzMeier, 2016-03-13:
Is this still open?

tubeme, 2016-03-13:
I use the default config with 1-4 quad; 5,6 elevons; 7 pusher, but with 400hz on all. No problem with digital servos. There will be problem with analog ones. So I solved this for myself, but still there is a flow if I have analog servos. Of course I might lower the 5,6,7,8 to 50hz in this case but then the pusher will work also on 50hz. 

I can close it?

LorenzMeier, 2016-03-24:
Closing this since the pusher at 50 Hz is perfectly appropriate.

closed 2016-03-24 by LorenzMeier