PX4/PX4-Autopilot #3926: VTOL transition altitude

sanderux, 2016-03-05:
Consistent problems with increasing altitude during transition:
http://logs.uaventure.com/view/NfbHqPWzcNNTaRE5KTechg
http://logs.uaventure.com/view/hMnRrAhpzmQmnYGaTZ94M4
http://logs.uaventure.com/view/UA7KfExFy7DAuvXt4z6TwQ

This was in +- 8m/s wind, airspeed sensor had spikes now and then.

sanderux, 2016-03-05:
/cc @AndreasAntener

sanderux, 2016-03-05:
[image]

Here's a plot of alt,  airspeed and pusher throttle. I think it is tecs that wants to convert kinetic energy to potential energy. @tumbili ?

AndreasAntener, 2016-03-05:
Duplicate: https://github.com/PX4/Firmware/issues/3843

AndreasAntener, 2016-03-05:
TECS is not active yet during the transition.

closed 2016-03-05 by AndreasAntener

sanderux, 2016-03-05:
Yep, just noticed that. any idea where this comes from?