PX4/PX4-Autopilot #3930: make posix_sitl_default jmavsim not working

pchatrath, 2016-03-05:
I get the following error output on running "make posix_sitl_default jmavsim"

```

-- posix-sitl-default
-- The CXX compiler identification is GNU 4.8.4
-- The C compiler identification is GNU 4.8.4
-- The ASM compiler identification is GNU
-- Found assembler: /usr/bin/cc
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Found PythonInterp: /usr/bin/python (found version "2.7.6") 
INCLUDE_DIRS=/home/pratik/src/Firmware/src;/home/pratik/src/Firmware/build_posix_sitl_default;/home/pratik/src/Firmware/build_posix_sitl_default/src;/home/pratik/src/Firmware/src/modules;/home/pratik/src/Firmware/src/include;/home/pratik/src/Firmware/src/lib;/home/pratik/src/Firmware/src/platforms;/home/pratik/src/Firmware/src/drivers/boards/sitl;/home/pratik/src/Firmware/build_posix_sitl_default;/home/pratik/src/Firmware/build_posix_sitl_default/src/modules/px4_messages;/home/pratik/src/Firmware/build_posix_sitl_default/src/modules;/home/pratik/src/Firmware/mavlink/include/mavlink;/home/pratik/src/Firmware/src/lib/DriverFramework/framework/include;src/lib/matrix;src/modules/systemlib;src/lib/eigen;src/platforms/posix/include;mavlink/include/mavlink
ADDING DRIVERS
CMake Error at CMakeLists.txt:355 (add_subdirectory):
  The source directory

    /home/pratik/src/Firmware/src/lib/ecl

  does not contain a CMakeLists.txt file.

CMake Warning (dev) at cmake/posix/px4_impl_posix.cmake:91 (get_target_property):
  Policy CMP0045 is not set: Error on non-existent target in
  get_target_property.  Run "cmake --help-policy CMP0045" for policy details.
  Use the cmake_policy command to set the policy and suppress this warning.

  get_target_property() called with non-existent target "lib__ecl".
Call Stack (most recent call first):
  src/firmware/posix/CMakeLists.txt:3 (px4_posix_generate_builtin_commands)
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Warning (dev) at cmake/posix/px4_impl_posix.cmake:91 (get_target_property):
  Policy CMP0045 is not set: Error on non-existent target in
  get_target_property.  Run "cmake --help-policy CMP0045" for policy details.
  Use the cmake_policy command to set the policy and suppress this warning.

  get_target_property() called with non-existent target "lib__ecl".
Call Stack (most recent call first):
  src/firmware/posix/CMakeLists.txt:3 (px4_posix_generate_builtin_commands)
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Warning (dev) at cmake/posix/px4_impl_posix.cmake:91 (get_target_property):
  Policy CMP0045 is not set: Error on non-existent target in
  get_target_property.  Run "cmake --help-policy CMP0045" for policy details.
  Use the cmake_policy command to set the policy and suppress this warning.

  get_target_property() called with non-existent target "lib__ecl".
Call Stack (most recent call first):
  src/firmware/posix/CMakeLists.txt:3 (px4_posix_generate_builtin_commands)
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Error at CMakeLists.txt:364 (add_subdirectory):
  add_subdirectory given source "src/lib/DriverFramework/framework/src" which
  is not an existing directory.

-- GIT_DESC = 07775f76296a8298a8cea3d33aece1f82e7c51d8
-- Configuring incomplete, errors occurred!
See also "/home/pratik/src/Firmware/build_posix_sitl_default/CMakeFiles/CMakeOutput.log".
make: *** [posix_sitl_default] Error 1

```

LorenzMeier, 2016-03-12:
Fixed on master.

closed 2016-03-12 by LorenzMeier

meeshawn, 2018-02-14:
Try the following before running "make posix_sitl_default jmavsim":
-git submodule update --init --recursive

ibrahimqazi, 2022-01-01:
> Try the following before running "make posix_sitl_default jmavsim": -git submodule update --init --recursive

I have the same issue, I already tried: `git submodule update --init --recursive` and then  
`make posix_sitl_default jmavsim` but I get:
```
make posix_sitl_default jmavsim
Makefile:496: *** "Make target posix_sitl_default not found. It either does not exist or posix_sitl_default cannot be the first argument. Use 'make help|list_config_targets' to get a list of all possible [configuration] targets.".  Stop.
```

julianoes, 2022-01-03:
@ibrahimqazi please don't necrobump all threads.

It's now `make px4_sitl jmavsim`.

ibrahimqazi, 2022-01-04:
> @ibrahimqazi please don't necrobump old threads.
> 
> It's now `make px4_sitl jmavsim`.

Thank you, I found it. In this PDF there is old command but on website the latest one is available.