PX4/PX4-Autopilot #3936: VTOL RTL with landing in FW lands diagonally

AndreasAntener, 2016-03-06:
because it will transition to MC somewhere along the loiter circle, then start to land at home. But unlike missions it won't first shift the landing location (home).

Solution to keep the way it currently works:
Add 2 new states in RTL after LOITER that will only be used for VTOLs:
- back transition
- move to land

A simpler solution, but changes the behaviour:
- 1 new state: back transition before loiter

Maybe we even want both options. And the user can select.

sanderux, 2016-03-07:
I think both loiter and land should be in MC mode. The default loiter radius in FW is dangerous as the area could be close to obstacles like buildings or trees.

AndreasAntener, 2016-03-07:
Sounds good, I think we can make that the default behavior.

We should even consider setting RTL_LAND_DELAY to 0 by default now so it actually does land. Because a) the landing works great now, and b) if it triggers RTL because of a real problem this usually isn't resolved while loitering at home (speaking from experience, and then you have to let your batteries run flat..).

sanderux, 2016-03-07:
Agreed, i'll get to work

AndreasAntener, 2016-03-23:
@sanderux FYI, I just implemented this, will make a PR soon.

sanderux, 2016-04-08:
@AndreasAntener i have the same behavior in the VTOL_LAND command. do you have a PR or shall i have a look at it?

AndreasAntener, 2016-04-08:
I'll open the PR this afternoon

closed 2016-04-21 by LorenzMeier