PX4/PX4-Autopilot #3937: High vibration alert during compass cal

DonLakeFlyer, 2016-03-06:
Would be nice if these alerts where turned off during calibrations which require moving the vehicle around.

kd0aij, 2016-03-06:
Hard to believe there's any "vibration" at all when doing a calibration.

tubeme, 2016-03-08:
I get "high vibration" warning every time while calibrating the compass between step 1 and 2, ie when I flip the plane upside down. Always! Probably ATT_VIBE_THRESH is too low by default. We regularly get high vibration warnings while flying as well without any noticeable vibrations (visually as well as in the Log graphics.)

mhkabir, 2016-03-08:
I get vibration warnings these days pretty randomly, even when just gently (smoothly) exciting the system in my hand. Did something change recently, because this didn't happen before.

LorenzMeier, 2016-05-22:
Fixed.

closed 2016-05-22 by LorenzMeier