PX4/PX4-Autopilot #3939: mavlink UDP broadcast doesn't work on Snapdragon

julianoes, 2016-03-06:
Currently, the UDP broadcast of a mavlink instance running on the Linux side on the Snapdragon fails.

This needs investigation.

liamstask, 2016-03-09:
cannot repro this locally (I'm on master @ `v1.1.3-686-gef8d074`), broadcast works as expected. if you're not connected successfully to a network, it may fail with `Network not reachable` but once connected it's all good.

julianoes, 2016-03-10:
@liamstask I've seen the same thing yesterday. Are you connecting to a Wifi or creating one on the Snapdragon?

liamstask, 2016-03-10:
snapdragon is in station mode via `/usr/local/qr-linux/wificonfig.sh -s station`, so connecting to an existing network.

julianoes, 2016-03-10:
@liamstask ok same setup for me.

julianoes, 2016-03-11:
In @bugobliterator 's case it doesn't work but that's using AP mode, so the Snapdragon is the access point.

LorenzMeier, 2016-03-24:
@julianoes Any news on this one, proper fix?

julianoes, 2016-03-24:
Raised this but have not gotten a solution yet.

julianoes, 2016-04-04:
Fixed in #4105.

closed 2016-04-04 by julianoes