PX4/PX4-Autopilot #3940: SENS_FLOW_ROT parameter does not work

priseborough, 2016-03-06:
The SENS_FLOW_ROT parameter used to allow for different installation angles for the PX4Flow sensor does not correct the published values.

mhkabir, 2016-03-07:
@priseborough That's weird. There is indeed a slight bug - it doesn't rotate the gyro readings, but the actual raw flow should be OK.

I fixed it here : https://github.com/mhkabir/Firmware/commit/b4e8e6eb678f7bde6963dc2be9d6b76e97c91f46

Will send in a pull request with that. Does that fix it for you?

priseborough, 2016-03-07:
Yes it does thanks.

mhkabir, 2016-03-08:
Done, applied via rebase.

priseborough, 2016-03-09:
The yaw rotation appears to be working for the values reported on mavlink, but does not appear to be applied to the values logged in the FLOW message. i will investigate further.

mhkabir, 2016-03-09:
That's weird. Mavlink and sdlog2 is grabbing the data from the same uorb topic which the driver publishes. (https://github.com/PX4/Firmware/blob/master/msg/optical_flow.msg)

priseborough, 2016-03-09:
This is on a different board, so it is possible the parameter is not being set properly. I have the board X-axis facing to the left (270 deg yaw) and have set the orientation parameter to 6. When the vehicle flies forward, the X-axis flow goes negative (the Y axis flow should go positive). 

I'll let you know if I find anything.

priseborough, 2016-03-09:
Changing the SENS_FLOW_ROT  parameter has no effect on either the mavlink or log output. Maybe i was mistaken about the mavlink output working before (the different line colors on QGC are not distinct)

mhkabir, 2016-03-09:
@priseborough Maybe yes. Could you just try master (verbatim) and check what works and what doesn't work for you?

priseborough, 2016-03-09:
OK when I loaded master, the flow data was being rotted correctly. I then rebased my branch on master and verified that the px4flow file was unaltered from master. I then repeated the test (leaving the parameter at 6) and the flow data was no longer being rotated. I then reset the parameter from its value of 6 back to zero with no change to the behaviour. I then set the parameter back to 6 and now it is working correctly. It appears as though the parameter was not setting correctly.

Anyway it appears as though the original issue has been resolved for master.

LorenzMeier, 2016-03-09:
The param might be lacking the restart required flag.

priseborough, 2016-03-09:
@LorenzMeier 

What appears to be happening in my ekf2 optical flow branch

https://github.com/priseborough/Firmware/tree/pr-optflowekf-rebased

is that when i restart, despite the SENS_FLOW_ROT reporting as 6 over mavlink, internally it is using the default of 0. If I use QGC to set the value back to 0 and then back to 6, it works. When I power cycle, although the parameter is still reported as being 6 over mavlink, internally it has gone back to using the default of 0.

Do you have any suggestions where i should start looking?

mhkabir, 2016-03-10:
@priseborough 
> despite the SENS_FLOW_ROT reporting as 6 over mavlink

What do you mean by "reporting over mavlink" ? You need to refresh your params in QGC in order to see the change, I believe.

priseborough, 2016-03-10:
yes, I did refresh parameters and even exited QGC and restarted it to be sure

mhkabir, 2016-03-10:
Something seems to be wrong with the parameter subsystem in that case. It looks to me that the param isn't being saved, just written. 

Basically you're saying that the param goes back to 0 after a power cycle, right?

priseborough, 2016-03-10:
That's the part i don't understand. The parameter reported (via refresh) after a power cycle is the value saved, however that value does not appear to be used. I have to set it to 0 and back to 6 again to get it to apply the yaw transformation. It is easy to test using the QGC real time plotting.

mhkabir, 2016-03-10:
I had a look at the code that sets the rotation here : 
https://github.com/PX4/Firmware/blob/master/src/modules/sensors/sensors.cpp#L904-L920

Maybe you can find something that I'm missing?

LorenzMeier, 2016-03-10:
The issue is likely your sensor Paul: 3DR equipped some modules with the wrong boot loader, leading to a very long boot time. That means that when the parameter gets set the sensor is not booted yet.

LorenzMeier, 2016-03-10:
To solve it we would need to move the param evaluation into the flow driver, which is architecturally undesirable but necessary.

LorenzMeier, 2016-03-14:
@priseborough I pushed a fix for this to master. Please re-test.

closed 2016-03-14 by LorenzMeier