PX4/PX4-Autopilot #3942: RTL uses MPC_XY_VEL_MAX

tops4u, 2016-03-07:
When activating RTL it seems to me that it will travel at maximum allowable Speed specified in MPC_XY_VEL_MAX. This may result in a huge overshoot of the Homelocation if other Parameters (I assume MPC_ACC_HOR_MAX, MPC_HOLD_MAX_XY) are not tuned accordingly. I like to have a fast Copter without the "twitch" of higher MPC_ACC_HOR_MAX / MPC_HOLD_MAX_XY Values. 

If RTL behave as described above a separate Value for the max allowed RTL Speed would be desirable.

AndreasAntener, 2016-03-07:
This will help once done: https://github.com/PX4/Firmware/pull/3567
You will be able to set a cruise speed next to max speed

tops4u, 2016-03-08:
@AndreasAntener will this also be Applicable for Mission Flights? Currently QGC only offers to set the Pitch Angle but it would be nice to set a CruiseSpeed for each Waypoint as well.

AndreasAntener, 2016-03-08:
This will be a general setting that applies to every position/velocity controlled mode including mission flight. A setting for each waypoint would be a different story.

tops4u, 2016-03-09:
@AndreasAntener  AFAIK currently QGC does not support setting a Speed Setting to fly to a Waypoint. This is not an Issue but Improvement. Shall I open an Issue on QGC? Would MAVLING support to submit a speed setting to reach a waypoint. Personally I would prefer having a RTL Speed and a separate Speed for each Waypoint Navigation in Mission Mode with a reasonable Preset in QGC.

AndreasAntener, 2016-03-09:
Setting speed on a waypoint is not supported by MAVLink. However, my answer above was incomplete in that regard. The PR I referred to will not only allow you to set a separate cruise speed (which will then apply for RTL), but also to specify "change speed commands". In QGC you should eventually be able to add such a command directly attached to a waypoint which will then result in what you need.

LorenzMeier, 2016-03-14:
You can set the cruise speed now on master separately from max speed.

closed 2016-03-14 by LorenzMeier