PX4/PX4-Autopilot #3943: New Simple Flight Mode Assignment bug

GervaisK, 2016-03-07:
With the new "Simple Flight Mode assignment" its very easy to assign modes to switch positions. However, PX4 doesn´t accept all choices. Example shown: Although Altitude Control is chosen and showing up below, it doesn´t appear in the headline and the copter doesn´t change to _Altitude_, it stays in _Position Control_.

[image]

Further: Users are confused, that chosen _Stabilize_ is shown as _Manual_ in the headline.

DonLakeFlyer, 2016-03-07:
The view in QGC (left arrow) is simulating flight mode selection based on channel values so you can test your settings. The firmware may in turn reject the mode change (the right arrow) so the toolbar may not change abd keep in sync. So these may not always be in sync.

GervaisK, 2016-03-07:
True and QGC doesn´t report the change as well as the Copter doesn´t. So I´m convinced that the mode did not change.

DonLakeFlyer, 2016-03-07:
QGC will only report flight modes changes by talking if the flight mode actually changes. It won't report simulated changes in Flight Mode config screen. As far as I can tell from the discussion things are working as they are supposed to. It may be confusing, but I'm not sure how else to show the user whether there flight mode settings are working other than simulating flight mode changes.

GervaisK, 2016-03-07:
Don, QGC is working correct. The New assignment doesn´t.

DonLakeFlyer, 2016-03-07:
Oh, sorry.

> Further: Users are confused, that chosen Stabilize is shown as Manual in the headline.

This is a QGC problem. It's trying to get fancy by changing naming based in multi-rotor/fixed wing. I don' think that will really work though. Can you put in a QGC issue for that.

GervaisK, 2016-03-07:
Yes,thx.

LorenzMeier, 2016-03-08:
Matching QGC issue: https://github.com/mavlink/qgroundcontrol/issues/2969

closed 2016-03-08 by LorenzMeier