PX4/PX4-Autopilot #3947: An error when i make px4fmu-v2_default in terminal with ubuntu.A newer needs help!

JarryChou, 2016-03-08:
in terminal:

wybbd@wybbd-virtual-machine:~/src/Firmware$ make px4fmu-v2_default
git submodule init
Tools/check_submodules.sh 
Git submodule config valid.
Checked NuttX submodule, correct version found
Checked Tools/gencpp submodule, correct version found
Checked Tools/genmsg submodule, correct version found
Checked Tools/jMAVSim submodule, correct version found
Checked Tools/sitl_gazebo submodule, correct version found
Checked cmake/cmake_hexagon submodule, correct version found
Checked mavlink/include/mavlink/v1.0 submodule, correct version found
Checked src/lib/DriverFramework submodule, correct version found
Checked src/lib/dspal submodule, correct version found
Checked src/lib/ecl submodule, correct version found
Checked src/lib/matrix submodule, correct version found
Checked src/modules/uavcan/libuavcan submodule, correct version found
Checked unittests/googletest submodule, correct version found
make -C /home/wybbd/src/Firmware/build_px4fmu-v2_default -j4 --no-print-directory 
make[1]: *** 没有指明目标并且找不到(no direction and could not find) makefile。 停止(stop)。
make: *** [px4fmu-v2_default] 错误(error) 2

kd0aij, 2016-03-08:
没有指明目标并且找不到(no direction and could not find) makefile。 停止(stop)。

translation?

JarryChou, 2016-03-08:
yes,it is

Inspirati, 2016-03-08:
output from:

*make --version*

and:

*cmake --version*

for example, mine looks like this:

$ make --version
GNU Make 4.0
Built for i686-pc-linux-gnu
Copyright (C) 1988-2013 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html
>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.

$ cmake --version
cmake version 3.4.1

CMake suite maintained and supported by Kitware (kitware.com/cmake).

mhkabir, 2016-03-08:
Just run a ```make clean``` and rebuild.

JarryChou, 2016-03-08:
thanks, it has been solved.

closed 2016-03-08 by sanderux