PX4/PX4-Autopilot #3949: PX4 UM: Cannot find Icons

tubeme, 2016-03-08:
Hi guys,

I'm writing the Users Manual and I need to find this icons:

[image]

I found the new style icons in the QGC github tree:

[image]

But for the setup explanation i need the ones with directions and motor numbers.

Any help will be appreciated.

dagar, 2017-04-11:
@tubeme there's now an effort to automatically generate some of this documentation. https://github.com/PX4/Firmware/pull/7019

closed 2017-04-11 by dagar

tubeme, 2017-04-11:
@dagar Thanks I follow the #7019