PX4/PX4-Autopilot #3950: PX4 UM: Parameters Refference

tubeme, 2016-03-08:
Hello,

I'm writing the UM of PX4 and I have the following problem. Somebody depreciated all the old docs and Index from pixhawk.org. The index page is still available but it is hidden. 

So it is a very good Idea if somebody from the core team can implement the Parameters reference page in the new Users Manual tree in the Advanced Configuration Section. 

https://pixhawk.org/firmware/parameters

It is automatically generated so I cannot create it. It is madness if I have to manually change the Parameters Reference Page when new parameters are added in the development process.

Thanks

dagar, 2017-04-11:
Fixed! https://dev.px4.io/parameter_reference.html

closed 2017-04-11 by dagar

tubeme, 2017-04-11:
@dagar Please can you make the Shortcut list on top of the parameters reference the same way it was in the original page?

dagar, 2017-04-11:
Check the little thing in the top right. It should expand into a table of contents.

tubeme, 2017-04-11:
Cool! :)