PX4/PX4-Autopilot #3952: MAVROS OFFBOARD command temporarily rejected: 176

sx00, 2016-03-08:
I got some useful information from these threads:
https://github.com/PX4/Firmware/issues/1288
https://github.com/mavlink/mavros/issues/180
https://groups.google.com/forum/#!topic/px4users/IC1iz_TsL_Y

but I still can't solve my problem.

I have quad with Pixhawk (firmware v.1.2.0) connected to RPi through USB.
I turn off airspeed, io_safety, and usb checkers.

I do the following:
1. start `roslaunch mavros px4.launch`. It's ok.
2. arm. It's ok.
3. then I try C++ examples from http://dev.px4.io/ros-mavros-offboard.html and https://pixhawk.org/dev/ros/mavros_offboard

In all cases I get this rejecting message - `command temporarily rejected: 176`

MarcGyongyosi, 2016-03-08:
Where are you getting your position information from? 

You shouldn't use USB as your connection - I think in recent versions it doesn't even let you fly with that connected by default.

sx00, 2016-03-08:
> Where are you getting your position information from?
I use it indoors - there are no satellites. Could this be a problem?

> You shouldn't use USB as your connection - I think in recent versions it doesn't even let you fly with that connected by default.
It's because i have unresolved issue with MAVROS serial port connection. I turned off USB checker.

Another thing I forgot to say - I use it without RC.

sx00, 2016-03-09:
`rostopic echo /mavros/local_position/local` has following output:

```
---
header: 
  seq: 14261
  stamp: 
    secs: 1457543694
    nsecs: 132650368
  frame_id: fcu
pose: 
  position: 
    x: 0.0
    y: 5.10771362109e-17
    z: 0.417076468468
  orientation: 
    x: -0.00240536188308
    y: 0.00170342885147
    z: -0.58965977228
    w: 0.807646374052
---
header: 
  seq: 14262
  stamp: 
    secs: 1457543694
    nsecs: 168650368
  frame_id: fcu
pose: 
  position: 
    x: 0.0
    y: 5.1150802399e-17
    z: 0.417677998543
  orientation: 
    x: -0.00240597918306
    y: 0.00170101390777
    z: -0.589648273809
    w: 0.807654772171
```

I tried to replace `mavros/setpoint_position/local` to `mavros/local_position/local`
in

```
ros::Publisher local_pos_pub = nh.advertise<geometry_msgs::PoseStamped>
            ("mavros/setpoint_position/local", 10);
```

but it doesn't work.

Main mavros output:

```
process[mavros-1]: started with pid [3448]
[ INFO] [1457543200.981782768]: FCU URL: /dev/ttyACM0:57600
[ INFO] [1457543200.984265008]: serial0: device: /dev/ttyACM0 @ 57600 bps
[ INFO] [1457543200.988141831]: GCS bridge disabled
[ INFO] [1457543201.035521675]: Plugin 3dr_radio blacklisted
[ INFO] [1457543201.035927612]: Plugin actuator_control blacklisted
[ INFO] [1457543201.338871310]: Plugin altitude loaded and initialized
[ INFO] [1457543201.895652612]: Plugin cam_imu_sync loaded and initialized
[ INFO] [1457543201.948158289]: Plugin command loaded and initialized
[ INFO] [1457543201.948502247]: Plugin distance_sensor blacklisted
[ INFO] [1457543202.023465372]: Plugin ftp loaded and initialized
[ INFO] [1457543202.110223966]: Plugin global_position loaded and initialized
[ INFO] [1457543202.110496153]: Plugin image_pub blacklisted
[ INFO] [1457543202.189140580]: Plugin imu_pub loaded and initialized
[ INFO] [1457543202.238789122]: Plugin local_position loaded and initialized
[ INFO] [1457543202.245813914]: Plugin manual_control loaded and initialized
[ INFO] [1457543202.292636726]: Plugin mocap_pose_estimate loaded and initialized
[ INFO] [1457543202.317698809]: Plugin param loaded and initialized
[ INFO] [1457543202.376021101]: Plugin px4flow loaded and initialized
[ INFO] [1457543202.421330216]: Plugin rc_io loaded and initialized
[ INFO] [1457543202.421617872]: Plugin safety_area blacklisted
[ INFO] [1457543202.462304487]: Plugin setpoint_accel loaded and initialized
[ INFO] [1457543202.462555997]: Plugin setpoint_attitude blacklisted
[ INFO] [1457543202.462718757]: Plugin setpoint_position blacklisted
[ INFO] [1457543202.514118705]: IMU: High resolution IMU detected!
[ INFO] [1457543202.597507664]: Plugin setpoint_raw loaded and initialized
[ INFO] [1457543202.597907299]: Plugin setpoint_velocity blacklisted
[ INFO] [1457543202.685298184]: Plugin sys_status loaded and initialized
[ INFO] [1457543202.731812924]: Plugin sys_time loaded and initialized
[ INFO] [1457543202.749463028]: Plugin vfr_hud loaded and initialized
[ INFO] [1457543202.750791153]: Plugin vibration blacklisted
[ INFO] [1457543202.846697195]: CON: Got HEARTBEAT, connected. FCU: PX4
[ INFO] [1457543202.848508861]: IMU: High resolution IMU detected!
[ INFO] [1457543202.887017038]: Plugin vision_pose_estimate loaded and initialized
[ INFO] [1457543202.940138653]: Plugin vision_speed_estimate loaded and initialized
[ INFO] [1457543202.994184955]: Plugin waypoint loaded and initialized
[ INFO] [1457543202.994724382]: Autostarting mavlink via USB on PX4
[ INFO] [1457543202.995087351]: Built-in SIMD instructions: None
[ INFO] [1457543202.995328288]: Built-in MAVLink package version: 2016.2.5
[ INFO] [1457543202.995567403]: Built-in MAVLink dialect: ardupilotmega
[ INFO] [1457543202.995771205]: MAVROS started. MY ID 1.240, TARGET ID 1.1
[ WARN] [1457543202.999421882]: TM: Clock skew detected (-1457541297.845008135 s). Hard syncing clocks.
[ INFO] [1457543203.860530371]: VER: 1.1: Capabilities 0x00000000000004eb
[ INFO] [1457543203.861067923]: VER: 1.1: Flight software:     00000000 (f29c7c4db6ba9627)
[ INFO] [1457543203.861643132]: VER: 1.1: Middleware software: 00000000 (f29c7c4db6ba9627)
[ INFO] [1457543203.862005007]: VER: 1.1: OS software:         00000000 (0000000000000000)
[ INFO] [1457543203.862383548]: VER: 1.1: Board hardware:      00000000
[ INFO] [1457543203.862842455]: VER: 1.1: VID/PID: 26ac:0011
[ INFO] [1457543203.863139955]: VER: 1.1: UID: 3333471335313431
[ INFO] [1457543789.602633273]: FCU: [cmd] ARMED by arm/disarm component command
[ INFO] [1457543789.654318013]: FCU: [blackbox] /fs/microsd/log/2016-03-08
[ INFO] [1457543789.703459679]: FCU: TAKEOFF DETECTED
[ INFO] [1457543790.060496502]: FCU: [blackbox] recording: 08_44_24.px4log
[ERROR] [1457543790.820878897]: FCU: Rejecting mode switch cmd
[ERROR] [1457543790.872136762]: FCU: command temporarily rejected: 176
[ WARN] [1457543795.840914468]: CMD: Unexpected command 176, result 1
[ERROR] [1457543795.855408218]: FCU: Rejecting mode switch cmd
[ERROR] [1457543795.905389937]: FCU: command temporarily rejected: 176
[ERROR] [1457543800.937147331]: FCU: Rejecting mode switch cmd
[ERROR] [1457543800.987923164]: FCU: command temporarily rejected: 176
[ INFO] [1457543805.256045454]: FCU: [cmd] DISARMED by arm/disarm component command
[ INFO] [1457543805.305754204]: FCU: LANDING DETECTED
[ WARN] [1457543805.938727590]: CMD: Unexpected command 176, result 1
[ERROR] [1457543805.967595610]: FCU: Rejecting mode switch cmd
[ERROR] [1457543806.018252642]: FCU: command temporarily rejected: 176
[ INFO] [1457543806.932975975]: FCU: [blackbox] recording stopped
[ERROR] [1457543811.004390713]: FCU: Rejecting mode switch cmd
[ERROR] [1457543811.055067640]: FCU: command temporarily rejected: 176

```

offb_node:

```
[ INFO] [1457543795.791876344]: Offboard enabled
[ INFO] [1457543800.840993425]: Offboard enabled
[ INFO] [1457543805.891590402]: Offboard enabled
[ INFO] [1457543810.941923786]: Offboard enabled
```

sx00, 2016-03-09:
Yes, I think that the problem is the position estimation. I will try with px4flow

sx00, 2016-03-09:
Suddenly I solved the problem **setpoint_position plugin and some others were blacklisted**

closed 2016-03-09 by sx00

saryee83, 2016-08-18:
@AleksandrKh Could you please give some details on how you solve this problem?  Many thanks!