PX4/PX4-Autopilot #3954: MC: PosCtl Landing not recognized

tops4u, 2016-03-09:
I'll explain by a first Hand example what the Problem is. I landed close to a House under a roof in PosCtl Mode (probably not a recomended way of operation). After landing (having Throttle in low position but not yet disarming) 4 of the 6 motors went out while the two front motors where continuing spinning on low speed. Suddendly (after 1-2 secs) the 2 Motors started spinning up very fast causing the Copter to tip over while landed - causing some minor mechanical damage. Further damage was avoided by flipping the Kill-Switch. FW used is 1.2 stable. 

My Explanation to this behaviour is that the HDOP of the GPS went from 0.6 to roughly 2 (as some of the Sats got shaded by Roof & House) causing the PosCtl Algorithm to assume a position deviation which it wanted to correct. IMHO when in PosCtl and 4 out of 6 Motors are off one might assume the vehicle landed. 

Could you clarify if this is a very bad way of operation, if Landing should have been detected or if in PosCtl and >50% of the Motors are off and Throttle is min the Vehicle shoud assume Landed Status?

Log file on uaventures is available.

LorenzMeier, 2016-03-09:
Your PWM_MIN setting is too low, the motors should never have shut off completely. You can't land in POSCTL like that - you need to switch to manual and disarm, or you need to start integrating something like a lidar and optical flow.

I'm closing this since its not a real issue but rather a mode of operation which is fundamentally dangerous and not safe (with any drone which doesn't have in-built lidar / ultrasound and optical flow).

closed 2016-03-09 by LorenzMeier

tops4u, 2016-03-09:
@LorenzMeier thanks. Coming from NAZA where this is a very usual scenario this should be included in the Docu.