PX4/PX4-Autopilot #3956: Snapdragon QGC connection problem

ChristophTobler, 2016-03-09:
The connection with qgc works at the beginning after starting the mainapp.

Content of mainapp.config:
uorb start
muorb start
mavlink start -u 14556
sleep 1
mavlink stream -u 14556 -s HIGHRES_IMU -r 50
mavlink stream -u 14556 -s ATTITUDE -r 50
mavlink boot_complete

Console output:
WARN  mode: 0, data rate: 2880 B/s on udp port 14556
WARN  offboard mission init: ERROR
INFO  Setting up UDP w/port 14556

WARN  stream HIGHRES_IMU on UDP port 14556 enabled with rate 50.0 Hz

WARN  stream ATTITUDE on UDP port 14556 enabled with rate 50.0 Hz

pxh> WARN  changing partner IP to: 192.168.1.225

After a few seconds I get the INFO (continuously):
INFO  Could not get lock, file name: /media/christoph/SSD/src/snapdragon/Firmware/src/platforms/posix/px4_layer/shmem_posix.c, line number: 227

I used the newest master version

LorenzMeier, 2016-03-09:
@julianoes @jywilson would be good if you could have a look.

tumbili, 2016-03-09:
@julianoes @mcharleb This can also be reproduced by e.g.
```
param show INAV*
```

julianoes, 2016-03-09:
Thanks for reporting, I'm on this.

kevinmehall, 2016-03-09:
The ioctls underlying the failing code seem rather strange. They [added non-standard ioctls to the standard Linux /dev/mem device](https://us.codeaurora.org/cgit/external/gigabyte/kernel/msm/commit/drivers/char/mem.c?h=LNX.LER.1.0-68040-8x74.0&id=7d81a69ae19427889a6464ded935afe457fddc69), which seem to do atomic operations on a hard-coded physical address. I'm not sure what part of that needs to be in kernel space, but it should be a new kernel driver or part of the adsprpc driver instead of tacked on to the standard `/dev/mem` device.

julianoes, 2016-03-10:
Note that we only have a workaround in place which means "no locking". Therefore let's keep this open until the ioctl is working properly.

julianoes, 2016-05-03:
Not resolved just yet, sorry. Still gotta track it down.

julianoes, 2016-07-13:
Fixed with #4859.

PR #3960 (2016-03-09): Fix shmem locking
This PR is to track the changes to resolve #3956.

Currently it should run even though the locking itself still fails.
closed 2016-07-13 by julianoes