PX4/PX4-Autopilot #3959: QGC Connection Issue

tumbili, 2016-03-09:
After flashing a new firmare (just a line change in the attitude estimator) I cannot connect to QGC anymore. This is what I get in the PX4 shell:

```
8270 HRTWorkQueue::process_trampoline
/home/cberg/Desktop/r00015/oe-core/build/tmp-eglibc/work/cortexa8hf-vfp-neon-linux-gnueabi/adsprpc/1.0-r0/adsprpc-1.0/src/fastrpc_apps_user.c:136:failed to create tls key/home/cberg/Desktop/r00015/oe-core/build/tmp-eglibc/work/cortexa8hf-vfp-neon-linux-gnueabi/adsprpc/1.0-r0/adsprpc-1.0/src/listener_android.c:112:listener using ion heap: -1
/home/cberg/Desktop/r00015/oe-core/build/tmp-eglibc/work/cortexa8hf-vfp-neon-linux-gnueabi/adsprpc/1.0-r0/adsprpc-1.0/src/fastrpc_apps_user.c:225::error: -5: 0 == (nErr = dlerr)
/home/cberg/Desktop/r00015/oe-core/build/tmp-eglibc/work/cortexa8hf-vfp-neon-linux-gnueabi/adsprpc/1.0-r0/adsprpc-1.0/src/fastrpc_apps_user.c:228:adsp dlopen error: px4muorb unrecognized file format /libmainapp.soERROR Initialize Error calling the uorb fastrpc initalize method..

WARN  mode: 0, data rate: 2880 B/s on udp port 14556
WARN  offboard mission init: ERROR
INFO  Setting up UDP w/port 14556

ERROR ERROR: FastRpcWrapper Not Initialized

WARN  stream HIGHRES_IMU on UDP port 14556 enabled with rate 50.0 Hz

WARN  stream ATTITUDE on UDP port 14556 enabled with rate 50.0 Hz

WARN  stream LOCAL_POSITION_NED on UDP port 14556 enabled with rate 50.0 Hz

pxh> WARN  changing partner IP to: 192.168.1.247
```

This line was not there before
```
ERROR ERROR: FastRpcWrapper Not Initialized
```

liamstask, 2016-03-09:
as of g9aab96a (with julian's aDSP image installed), I do not see `ERROR: FastRpcWrapper Not Initialized`. my output looks like:

```
/home/cberg/Desktop/r00015/oe-core/build/tmp-eglibc/work/cortexa8hf-vfp-neon-linux-gnueabi/adsprpc/1.0-r0/adsprpc-1.0/src/fastrpc_apps_user.c:136:failed to create tls key/home/cberg/Desktop/r00015/oe-core/build/tmp-eglibc/work/cortexa8hf-vfp-neon-linux-gnueabi/adsprpc/1.0-r0/adsprpc-1.0/src/listener_android.c:112:listener using ion heap: -1

WARN  mode: 0, data rate: 2880 B/s on udp port 14556
WARN  offboard mission init: ERROR
INFO  Setting up UDP w/port 14556

WARN  stream HIGHRES_IMU on UDP port 14556 enabled with rate 50.0 Hz

WARN  stream ATTITUDE on UDP port 14556 enabled with rate 50.0 Hz

pxh>
```

which makes me think this could possibly already be fixed in master.

LorenzMeier, 2016-03-24:
I think we covered this one.

closed 2016-03-24 by LorenzMeier