PX4/PX4-Autopilot #3964: Loiter mode not working

DonLakeFlyer, 2016-03-09:
* Using master SITL
* Create a simple takeoff and waypoint missing in QGC
* Send, Arm, Change to Mission flight mode
* Select Auto: Loiter from the flight mode drop down

Firmware says unknown flight mode

DonLakeFlyer, 2016-03-09:
Also see this: https://github.com/mavlink/qgroundcontrol/issues/2969

AndreasAntener, 2016-03-10:
@DonLakeFlyer I'm using QGC 2.9.4 and firmware 1.2.0 (a few weeks back). In QGC this is called "Auto: Pause" for me, is that new? And it's working. But I know the bug you encountered, had this too a while back but don't remember with what combination anymore.

DonLakeFlyer, 2016-03-10:
Yes "Auto: Pause" is Loiter. No idea when that changed! I don't really like the new Auto: foo naming of things. Doesn't really add much value.

LorenzMeier, 2016-03-10:
Its necessary to do it that way because Loiter keeps APM users thinking its position control mode.

tops4u, 2016-03-11:
@LorenzMeier Do I understand you correctly that in fact it would still be "Loiter" as in APM. If there is no mission present Auto Mission / Auto Pause would in fact loiter in place. If a Mission is present it would try to follow the mission but would be immediately be paused resulting in a Loiter as well. The Difference would be that if in Mission it would really pause/resume the Mission if switched directly to/from Mission Mode. If using any other mode it would reset the mission progress... Correct?

AndreasAntener, 2016-03-11:
@tops4u no, Loiter in APM is position control in PX4. And the mission progress is not reset when using any other mode.

AndreasAntener, 2016-03-11:
@DonLakeFlyer does it go into failsafe in your case (no joystick connected)? It could revert back to the mode `NAVIGATION_STATE_AUTO_RTGS` which would then show as unknown.

DonLakeFlyer, 2016-03-11:
Yes it does go into failsafe

LorenzMeier, 2016-03-13:
@DonLakeFlyer Fixed in c29bfcd and on QGC master (mode indication for RTGS).

closed 2016-03-13 by LorenzMeier