PX4/PX4-Autopilot #3966: Mission land sequence outputs strange messages

DonLakeFlyer, 2016-03-09:
Using master SITL
* Create simple takeoff/waypoint/land mission
* Start the mission and let it complete

After it lands, you get:
* Critical: mission finished loitering
* Critical: failsafe mode on
* Critical: no RTL when landed
* Critical: RTL: completed, landed
* You are left in Auto: Return to land mode

DonLakeFlyer, 2016-03-13:
More strangeness. If your mission ends at a waypoint (not a land item) it will still do an RTL after it hits the last waypoint!

LorenzMeier, 2016-03-13:
Only if you have no RC control, which is perfectly fine / correct. Use a joystick to avoid that.

AndreasAntener, 2016-03-13:
The message sequence is unfortunately the logic outcome of the states it goes through, I don't think that's a problem.
For your second issue, instead of having a joystick, enable data com loss: COM_DL_LOSS_EN

DonLakeFlyer, 2016-03-13:
> Only if you have no RC control, which is perfectly fine / correct. Use a joystick to avoid that.

How do I controll the vehicle with only a tablet?

AndreasAntener, 2016-03-14:
@DonLakeFlyer see my last comment, try COM_RC_IN_MODE = 1 and COM_DL_LOSS_EN = 1

LorenzMeier, 2016-03-14:
I have that already on master, closing.

closed 2016-03-14 by LorenzMeier