PX4/PX4-Autopilot #3967: Proposal: FLIGHT MODES

tubeme, 2016-03-10:
Hello everybody,

I'm thinking how to explain the modes in the users manual with icon graphics since last week. And I try find a way to explain it to a person that does not have any knowledge on drones. And believe me I cannot figure it out how to do it. I think the system will become monster with thousand of modes and sub modes and sub modes and yes for me that is very good but for the novice it is not.

There is no Acro mode for fixed wing in the users manual. Actually Manual in FW = Stabilized flight, while we have to switch to Acro in order to have real Manual Flight. 
Her is what it is now:

MANUAL
Fixed wing aircraft/ rovers / boats: The pilot's control inputs are passed directly to the output mixer.

Multirotors:

ACRO: The pilot's inputs are passed as roll, pitch, and yaw rate commands to the autopilot. This allows the multirotor to become completely inverted. Throttle is passed directly to the output mixer
RATTITUDE The pilot's inputs are passed as roll, pitch, and yaw rate commands to the autopilot if they are greater than the mode's threshold. If not the inputs are passed as roll and pitch angle commands and a yaw rate command. Throttle is passed directly to the output mixer.
ANGLE The pilot's inputs are passed as roll and pitch angle commands and a yaw rate command. Throttle is passed directly to the output mixer.

Which is not exactly what it is.

It is very perplexing and always overwhelming for the novice. So now my question is what is the default mode in Manual for the multicopters? If I don't program acro switch neither rat switch what is the mode i'm in? In FW I'm in Manual Stabilized flight if I don't have Acro switch.

I think that the current system of naming the modes is very contradictory thus hard to understand.

What is Assisted Flight? is it manual or auto mode?? In terms of user experience Assisted flight is Auto mode as well, because the computer automatically does things for me.

So when you have to understand the difference between Assisted mode and Auto mode you get confused because you perceive both as Auto modes, and they are in reality auto modes.

There should be simpler system for the modes convention and naming and explanation. I don't say change in the functioning and features but how they are called explained and experienced. Ease of use = larger user base. You can make a very complex system work in two ways, simple way and complex way depending on the user needs and experience.

So here is a simpler way of doing it:

Manual (without sub-mode switch, ie without Acro,Rat, Angle switch)

For all vehicles the default function should be Manual Stabalized. This way Manual = Stabalized and all sub-modes of Manual are not mandatory.

Assisted

In Assisted the default should be POSCTL by default. I never ever needed ALTCTL never used it nor i needed it. So ALTCTL can be invoked by missing GPS or other pos data, or if I have a sub-mode - posctl switch and I mandatory switch it to ALTCTL.

Mission

It is pretty explanatory. Just start mission execution.

AUTO_LOITER - should become just LOITER. It is my first line of defense, it is my yellow distress call. Especially if I'm beginner. I use it if I have to pause and take a breath in a situation, and then continue on. It is actually a PUSE button. So It should work in all modes! even in the Acro modes. (You can fly acro and still have GPS lock or pos data.)

AUTO_RTL should become RTL. It is my distress red call so you cannot make it as a sub-mode of Auto!!! Usually when i hit RTL I'm desperate... something is happening. It should work from every mode. Now it works from the assisted modes if GPS lock and from Auto mode and why is it then in Auto when it works from both Auto and Assisted?

That's it plain and simple. We have Mode switch with 3 modes Man / Ass / Miss, Manual switch, Assisted switch, Loiter Switch and RTL switch if I'm advanced.

If I'm novice I set Mode switch with 3 modes Man / Ass / Miss. And I can set up either RTL and Loiter switches or make an Emergency 3 pos switch with RTL in the middle position as a first line of defense and RTL in on position as a second line of defense.

This way as a novice I will have just two switches. And with this two switches with the default sub-modes I gave above I will have pretty much all that I need from my drone being FW, Multi or VTOL. For the VTOL we have just Transition switch.

And also the modes should be consistent throughout the vehicles.

Now for the VTOL they are not!!!

The main success of DJI is not their machines but their simple 3 MODES for flight and you notice it the moment you get the white minimalist RC.
GPS, Manual = Stabalized, , RTH. You need nothing more in 80-90% of the situations.

See their default mode is the GPS one! So we wait a little for the GPS lock before flight. If you have an urgent need to take off without GPS then you switch to Manual.

Of course Loiter is mandatory for the Airplanes. But for Multicopters you don't even need Loiter because POSCTL is the same as loiter in terms of dynamics. The only difference is the gimbals does not work in Loiter vs POSCTL.

I wrote this just to provoke discussion. I will start working on Flight Mode explanation in the Users Manual and will show you what I have in mind.

tubeme, 2016-03-10:
With the development of new features in the sub-modes it will be very easy to set the controls. Otherwise with time you have to create more and more and more switches and assigning them... crazy things for the novice. Even to some experienced people comming from other platforms.

So here is the convention.

**Mode** switch 3 pos Manual / Assisted / Mission

Manual Switch 2-3-4-5 postition you select how many positions and assign a sub-mode to it. You decide which ones. If there are 10 modes in manual no problem you choose the ones you want to use. The aasignment is done very very simple and elegant in DJI. We can borrow the idea. Here how it looks:

[image]

With a drop down you can assign a function to a segment of the scale. BTW you can see in reality how the slider is moving and where in real time. Very easy to trim and set.

**Assisted** Switch 2-3-4-5 position the same logic. for now just POSCTL and ALTCTL but in the future there might be more.

**Mission** Switch 2-3-4-5 the same logic.

**Emergency** switch - 3 pos No emergency / Loiter / RTL

**Transition** switch for the VTOL.

For the novice use just the Emergency and Mode switches. For the advanced users you can have a countless possibilities of combinations with just 5 switches, in an easy manageable WYSWYG way.

DonLakeFlyer, 2016-03-10:
Much nicer visual way to do 6 position simple mode flight mode setup. I like it.

tubeme, 2016-03-10:
Sorry for the mess. I first posted this in https://github.com/PX4/Devguide/issues/67
Then I created this issue.

I made mistake in the modes explanations for the DJI which @GervaisK corrected in this post:

https://github.com/PX4/Devguide/issues/67

Another provoking question.

Why I'm not able to arm in Acro or Assited mode?

tops4u, 2016-03-10:
@tubeme I used to fly a NAZA F450 prior to move to PX4 (on a Tarot 680), and I used to land and takeoff in any mode offered by the FC (including GPS and Alt). Taking off in Stab Mode is somewhat ok though I would prefer to be able to take off in PosCtl as well. Landing on the other hand is a bit shaky in Stabilized mode and I would also prefer to be able and land in PosCtl Mode. 
Despite in Auto Mission or RTL Mode take off an landing is possible, so why denying it to PosCtl? 
With the current state landing in PosCtl is a dangerous thing to do, as I documented here: #3954

GervaisK, 2016-03-10:
Thx for correction. As I already wrote @ #67, **AltCtl. is a mandatory FM **for me . 
The DJI FM selection shown above is not actually the latest (a bit better at A2 assistant) but gets really confusing when its coming to the IOC modes (Simple and super simple @ APM stack)
 
Secondly DJI is calling it Failsafe, because there is a selection between Land and RTH (RTL) at another page of the assistant. 

**So I like the way its done now at latest master**. Really superior to DJI, since this leaves the full freedom to choose the order (Switch Position) . DJI doesn´t offer this. (only for AUX @ A2)

**I´m only missing LAND as a choosable fixed Failsafe option**, because I know a lot of situations where I don´t need a climb (sts. dangerous) or GPS reception is very bad, just want to make sure that it doesn´t switch to RTL (when GPS Glitch is over).

And I´m missing LAND as an AUX function, where I´m able to terminate my testflight right away. It is not obvious if the FM AUTO/LAND is usable w/o a mission for that

tubeme, 2016-03-10:
@GervaisK IOC is the most stupid thing I ever encounter... I even don't mention it here. It is a failure in terms of features.

In the convention I proposed we can have many sub-modes in the Mission mode. They could be Land, Take off, Follow me, Circle facing a spot or Orbit, Panoramic Selfie etc. We could then decide which ones we like to have on the Mission dial.

tubeme, 2016-03-10:
[image]

tubeme, 2016-03-10:
[image]

tops4u, 2016-03-10:
I would not like to loose flexibility on the PWM Range used. Your Proposal looks to me to work on fixed PWM Ranges. If one User only has limited amount of RC Channels mixing different switches to one Channel is one option. This usually causes problems to 100% reach certain values. To be honest I am personally very happy with the flight mode Assignement as it was up to QGC 2.9.4. Your Proposal would limit the amount of configurable Flightmodes to 6 per Channel?

DonLakeFlyer, 2016-03-10:
I would not expose PWM range setting to the user (although it would be needed internally). I would just have them specify what type of switch they have: 2 position, 3 position, 6 position.

I’ve been thinking about this. I’ll send a picture from QGC sometime today folks can look at.

tubeme, 2016-03-10:
We are not loosing range. But we should take in mind that there are 2 conventions for the mid point. one is 1500us and the other one is 1520us which means that the scaling in both types of receivers is a little off.

GervaisK, 2016-03-10:
> I would not expose PWM range setting to the user
This is , at least in an advanced view, mandatory. 

If the PWM is too close to the next mode, the user doesn´t even notice and you can´t help in a forum if s. happens.
Once everybody starts to use a one for all DJI RC gear´w/o the possibility of adjusting PWM , it doesnt matter anymore...but for now.

DonLakeFlyer, 2016-03-10:
The live display on channel info should show that the switch position is too close to the edge of the range.

tubeme, 2016-03-10:
Actually I never had a DJI RC I was just giving an example with it. This slider is from NAZA and the only thing that you have to do in order to fix some misalignment is push a little the output scaling. This can be done easily with probably all RCs. And is straight forward procedure that you learn in ABC of drones. For the guys that buy let say a RTF product based on PX4 and with RC, everything will be preset for them in the factory. But once they have the knowledge and ability to make their own settigns and upgrade their RC it will be pretty easy.

I started writing this issue, because I want to explain the whole thing for Novice and for Experts. And when you start explaining things you see some patterns that are not logical for explanation.

I already introduced about 10 people to PX4 and every time I do it, I see their faces when I show them the 5 switches  (i never even set acro switch for them) with probably 3-4 combinations they have to learn in order to fly their machine.

tubeme, 2016-03-10:
With the risk of becoming annoying I will say the last thing I have in mind about the Modes.

I come from the world of marketing communication and positioning. I did projects for some popular brands like Loreal, Nestle, Absolut. So I come from the world of Marketing, Sales and Profits. That is the way I perceive the world around me.

In the marketing positioning game, there is a very important principle if you want to establish your positioning ( or called dimension sometimes) for a certain product or a brand. It is called the Ladder principle. Which means that in order to ingest some NEW information to somebody's brain, you have to use the knowledge they have already.

Also everything is communication not only your brand! The brand name is very important positioning statement but it is comparably little part of the communication. We have to remember that everything is establishing the positioning (or the dimension), your packaging (the way it looks, the way it feels, the way it smells, what material it is made of etc.), your Users Manual (how you communicate in it, how you treat your users in it, how you thank them for choosing your product etc.), your product User Experience ( how they use it, is it the way you think they are using it or it is something that you did not expect when you created it, is it easy, is it emotional etc.), your Customer Service, your reaction to crisis etc. All this creates the dimensions around your brand and establish your positioning.

So looking at the name PIXhawk at this angle it is a Killer brand!. Perfect communication, it means Digital Bird. The moment you read it you almost on the spot recognize this as a DRONE brand.

I cannot say this for the Modes though for now and they are very important part of the marketing communication because they are related to the user experience and the ease of use.

ALTCTL What is this? This is just a jargon acronym.

POSCTL - the same...

RTL

Loiter

all the same... While the RTL and Loiter could be known from some experience users it is not true for the AltCtl and PosCtl. I cannot relate them straight forward to a knowledge I already have. No way...

So if we have to make the right communication we should call them

Positition Control, Hold or just - **Position** mode

Alltitude Control, Hold or just - **Altitude** mode or **Assist** mode or **Attitude** mode

**Come Home**  instead of RTL

**Pause** instead of Loiter

**Acrobatic** mode or **Sport** Mode (the DJI convention). not just Acro.

**Mission**

**Selfie**

**Orbit**

**Follow me**

**Manual**

All these are good names because almost everybody knows the first meaning of this words, and then buying your product they will attach another meaning to this words ie you will build a dimension to your product.

IPhone is not the most profitable phone because it is the smartest phone. Actually it is the dumbest phone but it comes with one button, it shines, it is expensive, it makes the best photos 200fps video and it feels good in your hand and it is a statement. I cold not stand my iPhone 6 when I got it as present and I sold it after a month, but I'm probably 1% of the people that will do this and prefer a little smarter phone and like Android.

DJI is the iPhone of the industry. And despite i don't like DJI I fly it as the most reliable controller that can carry my camera (pretty expensive one) and they do their communication by the book. Once the Pixhawk and PX4 become the Android in the industry I will put my camera under it. There is a some work for us to achieve this. We are not there yet...

DonLakeFlyer, 2016-03-11:
Ok, thanks to reading through all of @tubeme comments. It got me thinking about a possible solution. Here is my clean slate proposal for flight modes:

First we need to think about what sorts of things can actually be assigned to a switch. After a while I decided there were only two types of thiings that can be assigned to a switch:
- Flight Mode - The vehicle has a set of flight modes, only one of which is active at a time.
- Functions - Each seperate function has two or more "modes" associated with it. For example: VTOL state is a function. It has two modes: Forward Flight, Hover. Another example is Landing Gear. It has two modes: Up, Down. Plenty of other possible functions out there to assign to a switch.

So you certainly need a Flight Mode switch and then you can have optional Function switches. But one single flight mode switch doesn't really cut it. There are too many modes to put them all on one switch. So you need the concept of Flight Mode "Override" switches. The flight modes you assign to these switches "Override" the setting on the main Flight Mode switch. Each Flight Mode has a priority, so if multiple overrides are active at the same time the highest priority one takes affect. So for example "Emergency Stop" would be the highest prioroty Flight Mode. Another important thing is that an Override switch always has to have an "Off" position, for when it is not overriding anything.

So how would you create a UI to make this work and can it be understand by users? In order to determine that I've prototyped this model in QGC. It's here in this branch: https://github.com/DonLakeFlyer/qgroundcontrol/tree/NewFlightMode. You can build it yourself if you want to play around with it. It works visually, but does not modfy params. In order to make this work firmware needs to change. The Default Configs correctly change thing. You can also manually tweak custom configs, but you may run into strangeness there.

Here is what it looks like:
[image]

At the very top right there is a dropdown which lets you pick from pre-configured options. We can add as many of these as wel like. I've built in 3 different one:
* Multi-Rotor with a 3 position main Flight Mode switch
* Multi-Rotor with a 6 position main Flight Mode switch
* VTOL with a 6 position main Flight Mode switch and Transition/Landing Gear function switches.

This image above is showing a 3 position main mode switch. It also has three override switches. The first the get you extra Loiter/Mission modes. The other two are what I call "oh shit" switches to RTL or Emergency Stop. In this example the Vehicle would be in Mission mode since that "overrides"  Alt Control on the main Flight Mode switch.

Here is a 6 position Flight Mode setup:
[image]
It's has the same flight mode selection capability as the first example:

And last here is a VTOL example, which uses Functions:
[image]
The concept of a Function is completely standalone from a Flight Mode.

dogmaphobic, 2016-03-11:
> With the risk of becoming annoying I will say the last thing I have in mind about the Modes.

@tubeme As far as I'm concerned, you're preaching to the choir :) I agree with all you say and I too get glaze eyed when I see things like *RATTITUDE*. And I write software for this thing!

The though part is reconciling the need of the novice, casual user and the desire of academia, a group who is just as much a target.

@DonLakeFlyer Nice gauges! :)

kd0aij, 2016-03-11:
here are 2 things going on here:

1. firmware needs an integer specifying the desired mode

DonLakeFlyer, 2016-03-11:
Novice picks a default config. We can create default configs that match common radios. Next step is to modify a default config and tweak it a bit. Power user still has every possible combination available as well.

tops4u, 2016-03-11:
@DonLakeFlyer I support your proposal, just one more thing - in an Advanced Mode or via Parameters it should still be possible to adjust the active PWM Range for each Mode activation. As written before please think of RC-Channels and not of Switches as People owning only small RC Systems with less than 8 channels (as I do) will use Switch Mixing to allow multiple PWM Settings on one RC Channel. On certain Systems it is fairly hard to adjust all Positions to specific PWM values. So the NAZA Aproach with unchangable PWM Ranges to activate certain Flightmodes would make this rather hard to configure. If this could be kept in mind when changing the assignement I think this would be a very comfortable way of configuring the RC Channel / Mode Assignement.

tubeme, 2016-03-11:
@DonLakeFlyer The new setup page looks very very promising. And this

>  Novice picks a default config. We can create default configs that match common radios. Next step is to modify a default config and tweak it a bit. Power user still has every possible combination available as well.

was exactly my point. We have a simple default config for the novice, that would work almost right out of the box, and then experienced users have all the necessary tools to make a more complex setups.

Also your idea about the Modes and Functions was exactly what i had in mind too. We can assign modes and functions the way we want. We just need the priority in order to have the right overrides during flight.

tubeme, 2016-03-11:
@DonLakeFlyer I think some sort of slider on the bottom only at the "us" current setting would be better than the blue ribbon  starting from 1000us. Or it could be a blue arrow kind of triangle not a slider.

LorenzMeier, 2016-03-13:
I'm looking at this today. Might need a call with Don to understand the details.

LorenzMeier, 2016-03-13:
I will have time to flight-test this with my team of the month. Thanks Don!

jcooper21, 2016-03-26:
I would like to add to this discussion a flight mode for fixed wings which is important. Something similar to APM planes FBWA mode. This should almost be the manual mode as a novice person has a high chance of crashing planes without pitch and roll being limited and a stall prevention active. If this already exists and I missed it then my apologies for the post.

DonLakeFlyer, 2016-03-26:
> I would like to add to this discussion a flight mode for fixed wings which is important.

You should put this in a separate issue

DonLakeFlyer, 2016-03-26:
@LorenzMeier Any news on what you want do do here?

marimach77, 2016-03-30:
@DonLakeFlayer I like your new flight modes concept very much 😁. Can't wait for official implementation in QGC & PX4 firmware. Any date?

DonLakeFlyer, 2016-04-01:
This is just a proposal at this point. Hasn't been approved.

tubeme, 2016-04-03:
@jcooper21 That mode you mention is what I'm looking for too for the novice. FBWA in my opinion is just like Manual - Stabalized but with stall prevention. 

http://ardupilot.org/plane/docs/fbwa-mode.html

tubeme, 2016-04-03:
If this is the new convention for the flight modes @DonLakeFlyer summarized in:

@value 0 Manual
@value 1 Altitude
@value 2 Position
@value 3 Mission
@value 4 Hold
@value 10 Takeoff
@value 11 Land
@value 5 Return
@value 6 Acro
@value 7 Offboard
@value 8 Stabilized
@value 9 Rattitude

I don't understand what will be the difference between Manual and Acrobatic?

I suggest that Rattitude should be implemented for FW flight as well and be named **3D Sport** mode for both MC and FW and VTOL in FW. In FW we will be able to fly stabilized, but just when we need it we can make 360's, rolls, loops etc and then when we need stabilization we could just release the sticks to just level the plane. It would be like gyro assisted 3D flying.

Probably Manual should mean Acrobatic. So **Acrobatic** is not needed, it is **Manual**.

AndreasAntener, 2016-04-04:
@tubeme `Acro` is actually different from `Manual` on multirotors, `Acro` is your pure 3d flying mode there.

Crazy idea: Maybe we should have `Stabilizied` as default mode for fixed-wing AND multirotor, and `Manual` would be manual for FW as we know it and you would have to assign a switch to get into `Manual`. `Acro` for multirotor would still remain the same as it is, because if you rename `Acro` to `Manual` people will start crashing their copters within a day. But there will be no `Manual` anymore for multirotors and `Stabilized` would actually mean the same for both (attitude stabilized)
What do you think of that?

I agree on `Rattitude`, although maybe just call it `Sport` which is neat and simple :)

Later, `Stabilized` in FW could evolve to something like @dagar 's airpseed mode or similar.

mhkabir, 2016-04-04:
@AndreasAntener That mode naming for multirotors sounds pretty nice. I had been meaning to bring that up sometime soon.

tubeme, 2016-04-04:
@AndreasAntener I see the problem with naming the modes arises because the current approach is to implement somehow similar modes for both types of vehicles the MC and FW. This way it becomes very confusing for the user, especially if he reads the current brief description of the modes: http://dev.px4.io/concept-flight-modes.html.

The approach I'm suggesting is to reverse the logic. See what each mode mean for the pilot of FW and MC independently. And name them correctly according to this logic.

You certainly got my point. **Manual** for **MC** means **Stabalized**. And **Manual** for **FW** means **Acro**. Not good idea especially for somebody is reading the manual. 
So your idea is great. Just remove Manual (it is not definite description, Manual has a different meanings for different people). Stabalized means one and only thing for everybody.

So **Stabalized** is the default mode for MC. **Stabalized** is the default mode for FW. And then we have the sport modes:

**Acrobatic** (full rate controll) (The meaning of acrobatic implies sport) for both FW and MC.
**3D Sport** (Ratitude) (3D gives better spacial orientation about the posibilities of flying, and it gives marketing spice to just Sport). So Marketing wise 3D Sport is perfect.

If you implement Ratitude for airplanes that would be the craziest signature mode ever. Although it is popular with the MC bunch, it will be new to the FW flying and establish a good following. If I have the ability to fly my plane 3D but with the comfort of having assistance here and there that will be killer mode.

There is another issue with the description of the modes. I got different approach in the new manual, then the current descriptions. I structured the presentation arround the vehicle not arround the modes. THis way the descriptions for FW, MC and VTOL are totaly separated so you cannot read what Manual is for the three vehicles in one paragraph and get confused.

I made it here: http://px4.io/?post_type=docit&p=1738&preview=true but it is still a draft that has to be approved and released by Lorenz.

Stabalized flight should be stabalized. I get confused by the @dagar Airspeed mode. It does not follow the classical way of controlling plane, so I have to try it and then to see if I have an oppinion. It sounds good, it has logic arround the beginner, but Stabalized is fundamental mode. It should stay as is. 

I would love Stall prevention in Stabalized. As I write about this stall detection and protection since 2 weeks two friend of mine crashed their Pixhawks with a stall in manual. One was begginer so when I asked him why he did not give enough throttle? He told me that he was afraid from the speed and not loosing his plane outside LOS, so he started flying it at 20% throttle to fly slow!!! I was stunned. But I realized that everybody that comes from a MC flying makes this mistake with the throttle. They think that if they decrease throttle they will slow things down, not realizing that we have a stall in the airplanes and this will crash your plane. The other guy was experienced, so he got in to conversation and just the plane slipped in stall, and neither he neither the PX4 managed to take it out of spin dive.

Because Lorenz just cut me on my suggestion to implement Stall detection and prevention in Stabalized i created this: https://github.com/PX4/Firmware/issues/4142 and I guarantee you it will be one of the most used modes here at my club.

Probably the argument is that the TECS are controlling the anti stall protection and they does not work in Stabalized, and the PX4 does not control the throttle in Stabalized. But I have 100 arguments that there are very simple things that can be implemented and work like a charm. For the real pilots Stall is the most important training you get during PPL course. And also having the stall warning is the N1 warning for all the pilots. Stall accounts to about 80% of the crashes of planes at my spot. The younger pilots crash everyday there with stalls. This is fundamental protection for FW flight it should be implemented everywhere, except Acrobatic Sport modes where you want to stall the plane on purpose and play with it.

tubeme, 2016-04-04:
All modes are **Manual** in a sense except Auto Modes. All the other modes you have some sort of manual input. Why use the term **Manual** then?

AndreasAntener, 2016-04-04:
> **Acrobatic** (full rate controll) (The meaning of acrobatic implies sport) for both FW and MC.

In my first attempt to respond here before I had the same thought. I have 2 issues I'd like to point out, both can be resolved:
- for FW this is not actually rate controlled, but "real" manual (you're right, that's a technicality, pls ignore me)
- VTOL: I would like to keep 2 separate **acro/manual** modes in MC and FW because although they are similar, a pilot who can fly FW in acro isn't necessary able to fly a MC in acro. So in the VTOL case I want FW acro but when I transition to MC I don't want MC acro (gets way to tricky for me ;)). This could be solved on a different level though for VTOL specifically, still worth to think about.

> I would love Stall prevention in Stabalized.

This is actually why I mentioned it, I think if we do it right we'd come up with a mode that has a very manual feel but helps you not stalling the plane.

> If you implement Ratitude for airplanes that would be the craziest signature mode ever

I'm pretty sure others have this as well but I would also find it very cool ;)

dagar, 2016-04-05:
Any volunteers for FW acro mode? I think all we'd need to do is add rate setters to ECL - https://github.com/PX4/ecl/blob/master/attitude_fw/ecl_controller.cpp#L123 and cleanup the calls in fw_att_control to set the rate directly based on stick input, and skip control_attitude. From there it's pretty easy to get to rattitude.

tubeme, 2016-04-05:
@AndreasAntener I see then what the problem is. The most confusing thing is using Manual as a label for a group of modes and at the same time having a mode with that name. So we have Manual mode and then Manual Acro and Manual Stabalized. That is how it is organized now. In the new convention if accepted there will be no sections. We will have just different modes and choose which one to assign to a switch. This way there will be no more confusion for the Manual Mode. It will appear in FW and MC air-frames with a slightly different meaning.

From this explanation:

> MANUAL
> Fixed wing aircraft/ rovers / boats: The pilot's control inputs are passed directly to the output mixer.
> Multirotors:
> ACRO: The pilot's inputs are passed as roll, pitch, and yaw rate commands [...]

I cannot understand what is the difference between Manual and stabalized in MC? Or is there stabalized in MC?

So that is how I understand it:

**MC**
**Manual** = Angle control, returns to level flight ie. stabalized.
**Acro** = can make 360 flip, does not return to level flight when sticks in the middle. Direct input to the mixer. 
**Ratitude** = Mixture between Manual and Acro depending on the sticks rate.

**FW**
**Manual** = direct commands to the mixer, direct response from the control surfaces and motor. Like you are flying it directly connected to the receiver. You can make 360 loops, when sticks are released the plane does not level itself. (Looks to me like analog to Acrobatic in MC)
**Stabalized** = Full manual control of the plane but in angle mode not directly to the mixer. Once the sticks are released the plane level itself. 

So there are the modes as of now and how I understand them.

I will have some thought about VTOL. Until now we very rarely use transition in other mode than Stabalized FW to stabalized MC as well as Altitude Control and Position control transitions.  So the average user will use mostly these three modes in my opinion. Probably even only Alt and Pos Control in general if there is good way for landing and taking off in Alt and Pos control.

tubeme, 2016-04-05:
In the current mode naming a transition from Manual MC to Stabalized FW is the most logical thing as dynamics but it is really confusing me from general knowledge.

Probably if the modes are:

MC 
**Manual** is renamed to **Stabilized**.
**Acrobatic**
**Ratitude** renamed to 3D Sport

FW
**Manual** is renamed to **Acrobatic**
**Stabalized**

This way the modes will function similarly in both MC and FW so will be easy to switch from MC to FW and vice versa.

tops4u, 2016-04-05:
@tubeme I was about to comment your earlier Post as confusing Names between FW and MC for Manual respectively Stabilized with different behaviour. 
I support your renaming proposal so that same Mode Names behave the same FW and MC. 
One may argue about Naming.
What is actuall called Acrobatic is called Manual in many popular Controllers (DJI, Tarot) but maybe it should not be reused after renaming the Flight Modes in order not to confuse existing users. 
Maybe it would be a good idea to keep Ratitude, APM uses "Sport" for something a bit different.

tubeme, 2016-04-05:
@tops4u I think that Naming the modes is a big part of the marketing communication chain. So Having signature mode names is part of it all. Of course we can follow all the trends in the naming but I think for the intermediate to expert transition to PX4 will not be a problem to switch in the mode naming. Otherwise we have to use Attitude for Stabilized; Sport for Acro etc. If we use Manual instead of Acrobatic then we could use Acrobatic instead of Ratitude ( 3D Sport ). Just it needs clean up once and for all.

LorenzMeier, 2016-04-17:
@tubeme Thanks Vasil! I think you're spot on with your observations and I expect to have more time to come back to this in the next few days.

LorenzMeier, 2016-05-22:
@tubeme @DonLakeFlyer Happy to pick the discussion up again now.

tubeme, 2016-05-22:
@LorenzMeier @DonLakeFlyer In the latest master if I don't have Acro switch programmed I was not able to get Stabilized mode except from the QGC interface. It is especially inconvenient for VTOL. 

For us the most used mode is Stabilized. In airplanes, in MC's and in VTOL. But we are not able to arm in Stabilized despite I perceive it as Manual mode.

Please explain to me in simple terms what is the difference what is the difference between Stabilized and Manual mode for MC?

And also what is the difference between Manual and Acro in FW?

What is the current Naming? I read somewhere @LorenzMeier explanation on the current new naming concept, but sadly could not remember where did i stumble on it.

dagar, 2016-05-22:
MC manual and stabilized are the same thing (rates and attitude).
FW acro and rattitude aren't implemented. FW Manual is full manual control like flying RC.

https://github.com/PX4/Firmware/blob/master/src/modules/commander/commander.cpp#L3292

tops4u, 2016-05-22:
@tubeme I can not confirm your observation. I did flights with Pixracer and AUAV-X2 today both having Master FW from 22.5. On both Stabilized worked and I was able to arm. On AUAV-X2 there is no ACRO Mode programmed as well since this is a Foto-Hexa. 

If you are not able to Arm there must be another Problem with your Configuration which prevents Arming.

tubeme, 2016-05-22:
On Apr 4 @AndreasAntener wrote:

> @tubeme Acro is actually different from Manual on multirotors, Acro is your pure 3d flying mode there.

Crazy idea: Maybe we should have Stabilizied as default mode for fixed-wing AND multirotor, and Manual would be manual for FW as we know it and you would have to assign a switch to get into Manual. Acro for multirotor would still remain the same as it is, because if you rename Acro to Manual people will start crashing their copters within a day. But there will be no Manual anymore for multirotors and Stabilized would actually mean the same for both (attitude stabilized)
What do you think of that?

I agree on Rattitude, although maybe just call it Sport which is neat and simple :)

Later, Stabilized in FW could evolve to something like @dagar 's airpseed mode or similar.

tubeme, 2016-05-22:
@tops4u Probably my master is a month old. It could be VTOL related will try tomorrow.

tubeme, 2016-05-22:
@dagar then is it logical to say this?

## MC 
**Manual** could be renamed to **Stabilized**.
**Acrobatic**
**Ratitude** renamed to **3D Sport**

## FW
**Manual** could be renamed to **Acrobatic**
**Stabalized**

Is correct

OR we could have the following:

## MC 
**Stabilized**.
**Acrobatic** could be renamed to **Manual**
**Ratitude** renamed to **3D Sport**

## FW
**Stabalized**
**Manual** 

Or even this:

## MC 
**Stabilized**.
**Acrobatic**
**Ratitude** renamed to **3D Sport**

## FW
**Stabalized**
**Manual** could be renamed to **Acrobatic**.

And it will still hold its logic right?

tubeme, 2016-05-22:
Manual in FW flight I understand it as if I connected the servos directly to the receiver right?

tubeme, 2016-05-22:
We need consistency between the modes because of the VTOL!!

If I make the transition with the VTOL the mode switch stays at the same position in both modes.

So The pilot expects consistent behavior.

So when I'm in Stabalized MC after the transition I want to be in Stabalized in FW.

If I'm very good pilot and I fly Acro MC after the transition I expect to be in Acro FW which is now called manual in FW.

From Altitude mode MC I want to go in Altitude mode FW.

So the consistency is very important for VTOL but it is important for people switching UAVs. Like people that fly Helicopters, MC's, airplanes as well. It will be much more easier to adopt the technology by pilots for all air frames. This is important for commercial use. And much more easier for explanation.

dagar, 2016-05-28:
There are some good ideas here. Can we try to gather our thoughts in a google doc and establish a path forward? This includes potential short term fixes (mode renaming) and some things that would be medium to longer term.

@tubeme @LorenzMeier @tops4u
https://docs.google.com/document/d/1pi7YMKu9dhcq6HRAXQ_gMesn2gGrRa7cdNTYLEsNDjE/edit

LorenzMeier, 2016-07-29:
Release 1.5 will handle flight mode cleanup.

closed 2018-02-21 by PX4BuildBot