PX4/PX4-Autopilot #3972: Implement MAV_CMD_DO_REPOSITION

dagar, 2016-03-10:
Now that QGC has guided mode support (https://github.com/mavlink/qgroundcontrol/pull/2979) I'd like to add MAV_CMD_DO_REPOSITION in PX4, but I'd like to work out the details first.

On the GCS side I think it makes sense to have a GUIDED mode you enter that then allows you to click on the screen to send MAV_CMD_DO_REPOSITION. @LorenzMeier what would you like to see here?

DonLakeFlyer, 2016-03-10:
For a fully mavlink generic implementation of all guided commands, here is what is needed:
- Mavlink command to goto and/or change altitude. MAV_CMD_DO_REPOSITION works for this although it needs to be spec'ed such that you can NaN all fields.
- There need to be mavlink commands for: Pause, RTL, Land, Takeoff. These cannot be done with flight mode changes commanded by QGC. The reason is that flight modes are opaque to QGC. So it doesn't know which flight mode is which. If Pause is going to be Pause/Continue then there needs to be status for Pause state so QGC knows which one to call.
- If there is to be a "guided" mode. Then that needs to be a mavlink command as well. There also need to be some mavlink status such that QGC knows when a Vehicle is in "guided" mode or not.

LorenzMeier, 2016-03-13:
@dagar It would be great if you could drive this. I would start minimal and just implement the MAV_CMD_OVERRIDE_GOTO (which is already deployed). This should result into a mode change to LOITER and the LOITER position should be updated using the coordinates.

You need to extend the mode switch logic here to switch to AUTO_LOITER:
https://github.com/PX4/Firmware/blob/master/src/modules/commander/commander.cpp#L847-L856

And here to copy the coordinates:
https://github.com/PX4/Firmware/blob/master/src/modules/navigator/navigator_main.cpp#L404

NOTE: Change from reposition to GOTO

something like:
```C++
_loiter->set_loiter_coordinate(lat, lon, alt);
```

Make sure to check that if the mode switch to loiter nav mode comes later the loiter coordinate is not overwritten (you could set a flag that it was set externally and not overwrite if this is the first mode switch).

dagar, 2016-03-15:
Looking at this now. What's the difference between MAV_GOTO_DO_HOLD and MAV_GOTO_HOLD_AT_CURRENT_POSITION?

Is there any need for MAV_CMD_DO_REPOSITION?

LorenzMeier, 2016-04-23:
@dagar I've started to work on the reposition branch on this. I intend to squeeze this into the 1.3 release because 1.4 will see more architectural improvements and I want to have low-risk functionality changes in before we start moving on that.

dagar, 2016-04-23:
Sounds good. This stalled for me because I wasn't sure how to proceed on the QGC side. How should you enter the state that allows clicking on the screen to send the cmd, etc.

LorenzMeier, 2016-04-23:
Closing this in favour of https://github.com/PX4/Firmware/pull/4347

closed 2016-04-23 by LorenzMeier