PX4/PX4-Autopilot #3977: AUTO Modes - no Sticks input possible

sn0west, 2016-03-11:
In MISSION, LAND, RTL, PAUSE Modes, stick inputs are not accepted by the multicoper. 
In my opinion this is dangerous and should be similar to the apm stack. 

I'm alway ready to switch to an other mode. But still I think it should be possible to adjust the position without leaving the mode. Eg. RTL, while sinking the copter needs some adjustment otherwise it will land on my car and not on the road. Or in MISSION PAUSE I want to adjust the position  to get a better shot with my cam. 
I don't want to actively fly in those Modes, I just want to be able to adjust the copter in the air befor it land.

ecmnet, 2016-03-12:
This, in my opinion, also applies at least to pitch/roll control in offboard-mode (if RC is available).

sn0west, 2016-03-13:
Any reflections on this issue? What do you think about being able to correct the autonomous mission in flight.

LorenzMeier, 2016-03-13:
I think we'll go with your recommendation and allow inputs. But it will take some time to bring these in.

ecmnet, 2016-03-14:
@sn0west Maybe there are two aspects regarding this issue. One is to correct autonomous mission and to continue. The other might be just to abort those missions/offboard control, when sticks are moved and return to a more controllable mode (e.g. poshold).

sn0west, 2016-03-14:
In my opinion is the first aspect the more important one. Correct a mission and let it continue.
Aboarding is always an option, but this works now already, just switch to stabilized. Or do I misunderstand you in your second aspect?

ecmnet, 2016-03-14:
@sn0west You are right, using a switch to abort is already an option. Touching sticks would just shorten reaction time in case of dangerous situations.

whyscience, 2016-04-25:
I am also working on this .
My solution is when there is RC input ,navigation_mode will turn into Position control. 
after this it will be land/rtl again.
It works , but more details need  to be  done.

dagar, 2016-04-25:
The part I'm wondering about is if you'd actually want stick input to cause the mode to change or just tweak the current action.
Eg my plane is on final approach for a landing waypoint and looks like it might be coming uncomfortably close to my car. Can I just adjust the heading slightly rather than abruptly leave the mode and reenter.

sn0west, 2016-04-25:
In my case it's only a little tweak. Not only on rtl and land but also on loiter (hold) Would be nice too.

jcooper21, 2016-04-30:
I think nudging is very important for any aircraft. Even Dji allows full throttle nudging, pitch and roll in every mode. Nudging is very important especially when things start to go bad. The less switches you have to hit to get control the better.

LorenzMeier, 2016-07-29:
Release 1.5 will handle flight mode cleanup.

LorenzMeier, 2017-04-10:
This has been changed so you can break out of auto.

closed 2017-04-10 by LorenzMeier

sn0west, 2017-04-10:
Great news! What version then? Do you need some feedbacks or testings?

mhkabir, 2017-04-10:
So I think at the moment, we are just kicked out of AUTO if the sticks are moved, as opposed to allowing an elastic path change (and continue on mission after correction) as most people want.

@LorenzMeier should we reopen this?

LorenzMeier, 2017-04-10:
We'll see about what user mode makes sense. Right now my first preference is to switch the mode but allow to continue the mission. The elastic path change gets us into a boat load of other issues that would require special handling, e.g. refactoring how waypoints are considered as reached. I think we should do that when Bezier curves have been added by @Stifael and not before.

mhkabir, 2017-04-10:
Sounds good.