PX4/PX4-Autopilot #3978: AUTO MISSION - only one time possible

sn0west, 2016-03-11:
A mission can only be activated once. Why? 
I do activate it by switch and when the mission is finished with a land, I desarm the copter, switch back to stabilized, arm it and want to start the mission but it only take off and stay in pause mode. Why?

sn0west, 2016-03-13:
Normal or not? Anyone on this?

dagar, 2016-03-13:
It's not just you. In the next version of QGroundControl you'll be able to click a waypoint to restart the mission.
When would you like the mission to reset?

DonLakeFlyer, 2016-03-13:
I think it should reset after you've landed.

sn0west, 2016-03-13:
The copter doesn't desarm, after landing. So I don't know what makes more sense. To desarm after landing and the reset the mission or after landing. But then you actually still in thr mission. So in my opinion it's better to disarm after landing and then reset it.

AndreasAntener, 2016-03-13:
I'd suggest we reset when disarming, a mission could include a landing.

DonLakeFlyer, 2016-03-13:
> The copter doesn't desarm, after landing.

There is a parameter to turn that on. It's is not currently the default, but it should be.

LorenzMeier, 2016-03-13:
@AndreasAntener Could you look into this?

AndreasAntener, 2016-03-27:
Fixed on master, mission reset when disarming.

closed 2016-03-27 by AndreasAntener