PX4/PX4-Autopilot #3987: Flight Mode, RC Channel parameter meta data cleanup

DonLakeFlyer, 2016-03-11:
* Flight mode names should not be in "ALL CAPS". 
* "Rc Channel ##" seems a bit noisy as well. Why not just "Channel ##", or even just "##"?

As short as possible, but still conferring correct meaning to user is key here. The reason is that long strings for these causes issues on small displays like phones. So whatever can be done would be appreciated.

Once the naming is worked out, QGC needs to change to be consistent with flight mode naming for the flight mode drop down.

Related to these QGC issues: 
https://github.com/mavlink/qgroundcontrol/issues/2993
https://github.com/mavlink/qgroundcontrol/issues/2969

dagar, 2016-03-12:
Ideas:
AUTO:PAUSE -> Hold
AUTO:RTL -> Home
AUTO:MISSION -> Nav
ALTCTL -> Altitude
POSCTL -> for fixedwing position doesn't really make sense, maybe Heading or Track

DonLakeFlyer, 2016-03-12:
I'd try to be as close to ArduPilot as possible since there are a decent number of people coming across now. Which means the main confusing is around Loiter/PosCtl.

LorenzMeier, 2016-03-12:
There are more DJI users and we want to steal from their user base, not from APMs.

LorenzMeier, 2016-03-12:
@DonLakeFlyer Changed on master to:
```
 * @value 0 Manual
 * @value 1 Altitude
 * @value 2 Position
 * @value 3 Mission
 * @value 4 Hold
 * @value 10 Takeoff
 * @value 11 Land
 * @value 5 Return
 * @value 6 Acro
 * @value 7 Offboard
 * @value 8 Stabilized
 * @value 9 Rattitude
```
Please close this once you have the renaming in QGC In.

LorenzMeier, 2016-03-12:
Ah and I shortened the channels.

DonLakeFlyer, 2016-03-12:
Awesome. I'll update QGC. Great point on DJI.

DonLakeFlyer, 2016-03-12:
Also did you bump to minor version number so it will override older meta data?

LorenzMeier, 2016-03-13:
I just did on master. No pull against QGC in yet. Probably makes most sense if you do this.

closed 2016-03-13 by LorenzMeier

DonLakeFlyer, 2016-03-13:
The minor version number bump is in the firmware parameter python parser, not QGC. Whenever you have parameter meta data changes which you want to override previous meta data you need to bump that minor number.

LorenzMeier, 2016-03-13:
And that's what I did. Someone still needs to copy the file across, because your primary user base is using Android, and those don't get it from the Firmware.

DonLakeFlyer, 2016-03-13:
Ah, yeah. Good point. We need to change that so meta data can download from the web based on vehicle settings.