PX4/PX4-Autopilot #3991: ALTCTL, POSCTL Throttle and Altitude.

tubeme, 2016-03-12:
Hello,

What is the latest functioning of the throttle in the Assisted modes now? IE is the throttle centered at 1500us and at this throttle the plane is holding the altitude. With the increase of throttle you gain altitude slowly, with the decrease you loose altitude slowly. Like in the MC.

Also if I pitch up at centered throttle will I gain alt? and vice versa pitch down will dive?

If throttle at 1000us ie at zero will the motor stop or idle?

dagar, 2016-03-12:
Throttle is airspeed min - max. Neutral (1500) is trim airspeed. If landed and 0 throttle then the motor will stop as a safety feature.

Now that I think of it switching the Throttle and Pitch sticks in FW ALTCTL/POSCTL would probably make more sense and be consistent with MC.

tubeme, 2016-03-13:
I've seen different variants of the handling in POSCTL and ALTCTL during development. 

Can some of the core devs kind of summarize all the variants for the throttle and altitude handling tried in ALTCTL and POSCTL during development. It seems to me that the only difference between these two modes now is the course handling. IE keeping course and alt vs just keeping altitude.

What is the strategy if I want to rise or lower altitude? how can I do it automatically and gently in this modes? Instinctively I will use the pitch command for this but what will happen with the throttle at middle or in some extreme points when the pitch is up or down?

tubeme, 2016-03-13:
Thanks @dagar 

I've seen different implementations of the throttle and pitch behavior in ALTCTL and POSCLT during development.

What is the strategy for gaining or decreasing altitude in a gentle manner? Let say if I want to increase altitude gently I will use pitch command. What is going on with the throttle at this time? If the throttle is at middle, or extreme position?

It seems to me that now the only difference between ALTCTL and POSCTL is the straight line coarse vs. just holding altitude.

Can somebody from the core devs summurize what variants were used for the pitch and throttle in ALTCTL and POSCTL?

Thanks

dagar, 2016-03-13:
The code is pretty readable if you're curious.
https://github.com/PX4/Firmware/blob/master/src/modules/fw_pos_control_l1/fw_pos_control_l1_main.cpp#L1087

The throttle stick is just an airspeed setpoint. FW_AIRSPEED_MIN to FW_AIRSPEED_MAX. The actual throttle of your plane is going to be set by TECS trying to maintain the airspeed you set with your throttle stick while changing or maintaining altitude based on your pitch stick.

tubeme, 2016-03-14:
Thanks @dagar So i can gently gain or loose speed. with the pitch while TECS will maintain my airspeed. Sounds good to me. And it is easy to explain.

closed 2016-03-16 by tubeme