PX4/PX4-Autopilot #3996: param @unit cleanup

dagar, 2016-03-13:
I'm going to review all param @unit usage to enable QGroundControl unit conversions. https://github.com/mavlink/qgroundcontrol/pull/3007

@DonLakeFlyer @LorenzMeier  Let's discuss what we'd like to use because there's a lot of inconsistency.

degree/degrees/degrees * 1e7
m/meter/meters
s/second/seconds/ms/milliseconds/us
m/s/s m/s^2
Hz 1/s

DonLakeFlyer, 2016-03-13:
deg, m, sec, ms, us, m/s/s, hz, %

You could mark @unit boolean and I can pick that up and turn it into a an enum on the QGC side.

dagar, 2016-03-13:
Thanks, I'll start doing this

     @unit boolean

equivalent to

     @unit enum
     @value 0 Disabled
     @value 1 Enabled
     @min 0
     @max 1

LorenzMeier, 2016-03-14:
Closing this as the pull is in. Please send more PRs if you find more stuff.

closed 2016-03-14 by LorenzMeier